PINTO0309/Tensorflow-bin #1: stop working

nafitriaulia, 2018-11-25:
i tried to install tensorflow in raspberry pi 3, but when build parameter" sudo bazel build --config ...." the process stop working while process on 4233/8000s.

what should i do?

thank you

PINTO0309, 2018-11-25:
@nafitriaulia 

There are a few things I have to ask you a question.
Please answer all questions.
Otherwise I can not solve it.

1. Do you have a reason not to use ".whl"?
2. The version you are trying to build is "Tensorflow v1.11.0"? Or is it "Tensorflow v1.12.0"?
3. You need to extend the SWAP area to 2 GB, did you do it?
```bash
$ sudo nano /etc/dphys-swapfile
CONF_SWAPSIZE=2048

$ sudo /etc/init.d/dphys-swapfile restart swapon -s
```
4. What did you specify for the parameters in the "./configure" script?
5. Can you post the full error message because there are too few information you provided?

nafitriaulia, 2018-11-26:
@PINTO0309 

i followed every single instruction in this post

1. I use .whl exactly like this site. i use tensorflow-1.11.0-cp35-cp35m-linux_armv7l.whl https://github.com/PINTO0309/Tensorflow-bin/raw/master/tensorflow-1.11.0-cp35-cp35m-linux_armv7l_jemalloc.whl

2. it is tensorflow-1.11.0

3. but i have 22gb available sd card. i use 32gb sd card

4. Please specify the location of python. [Default is /usr/bin/python]: 

Found possible Python library paths:
  /usr/local/lib/python2.7/dist-packages
  /usr/local/lib
  /home/pi/tensorflow/tensorflow/contrib/lite/tools/make/gen/rpi_armv7l/lib
  /usr/lib/python2.7/dist-packages
  /opt/movidius/caffe/python
Please input the desired Python library path to use.  Default is [/usr/local/lib/python2.7/dist-packages]

Do you wish to build TensorFlow with XLA JIT support? [y/N]: n
No XLA JIT support will be enabled for TensorFlow.

Do you wish to build TensorFlow with OpenCL SYCL support? [y/N]: n
No OpenCL SYCL support will be enabled for TensorFlow.

Do you wish to build TensorFlow with CUDA support? [y/N]: n
No CUDA support will be enabled for TensorFlow.

Do you wish to download a fresh release of clang? (Experimental) [y/N]: n
Clang will not be downloaded.

Do you wish to build TensorFlow with MPI support? [y/N]: n
No MPI support will be enabled for TensorFlow.

Please specify optimization flags to use during compilation when bazel option "--config=opt" is specified [Default is -march=native]: 

Would you like to interactively configure ./WORKSPACE for Android builds? [y/N]: n

5. there's no error message because my raspberry pi 3 stop working anything. it looks like "lag" but can not do anything, even moving the pointer mouse. so i should unplug the power cable and plug in again to restart the raspberry pi

PINTO0309, 2018-11-26:
@nafitriaulia 

>I use .whl exactly like this site. i use tensorflow-1.11.0-cp35-cp35m-linux_armv7l.whl 

After executing the following command, installation of "Tensorflow v1.11.0" and "TensorflowLite v1.11.0 " has already been completed.
You do not need to build Tensorflow.
```bash
$ sudo apt-get install python-pip python3-pip python3-scipy libhdf5-dev
$ sudo apt-get install -y openmpi-bin libopenmpi-dev
$ sudo pip3 uninstall tensorflow
$ wget -O tensorflow-1.11.0-cp35-cp35m-linux_armv7l.whl https://github.com/PINTO0309/Tensorflow-bin/raw/master/tensorflow-1.11.0-cp35-cp35m-linux_armv7l_jemalloc.whl
$ sudo pip3 install tensorflow-1.11.0-cp35-cp35m-linux_armv7l.whl
```

>but i have 22gb available sd card. i use 32gb sd card

The extension of the SWAP area must be set to eliminate RAM shortage.
It is necessary to allocate the SWAP area separately from the whole size of the SD card.
About 3 GB of RAM is the minimum required to compile Tensorflow, but RaspberryPi's RAM has only 1 GB.
It is a move to evacuate the portion which can not fit in RAM to SDCard.

>there's no error message because my raspberry pi 3 stop working anything. it looks like "lag" but can not do anything, even moving the pointer mouse. so i should unplug the power cable and plug in again to restart the raspberry pi

When SSH is used, it hangs up frequently in the middle of processing.
It is recommended to connect the RaspberryPi directly to the HDMI display.
When I was using SSH, I was repeating the same phenomenon as you.
SSH hangs up at high load.

nafitriaulia, 2018-11-26:
@PINTO0309 

thank you for your response.

oh i think i should build parameter also after do this step "sudo pip3 install tensorflow-1.11.0-cp35-cp35m-linux_armv7l.whl"

btw do you know solution on my issue? why did i still fail when execute tensorflow lite?

error message is
Traceback (most recent call last):
  File "tflitevideo.py", line 26, in <module>
    interpreter = tf.contrib.lite.Interpreter(model_path='quantized_model_cityscapesv3_2.tflite')
  File "/usr/local/lib/python3.5/dist-packages/tensorflow/contrib/lite/python/interpreter.py", line 53, in __init__
    model_path))
ValueError: Op builtin_code out or range: 97. Are you using old TFLite binary with newer model?Registration failed.

PINTO0309, 2018-11-26:
@nafitriaulia 

>Are you using old TFLite binary with newer model?

Probably this error message represents the opposite meaning.
Please try to regenerate the .tflite file from the .pb file using toco of v1.11.0.

nafitriaulia, 2018-11-29:
@PINTO0309 

Ok i will try. thank you for your advice

can i ask another thing?
I used tensorflow for object segmentation with video input. But raspberry stop running the program around 2 until 4 minutes. What do you think about it?  is it raspberry's fault or tensorflow's fault?

My program successfully running on my PC untill the end of video.

Thank you

PINTO0309, 2018-11-29:
@nafitriaulia 
I do not know the program you wrote at all, so I can not answer.
Rather than discuss here, posting a question with StackOverflow can clarify the solution.

PINTO0309, 2018-12-12:
Since two weeks have passed since the last reply, I will close the issue.

closed 2018-12-12 by PINTO0309