PINTO0309/Tensorflow-bin #2: it hangs after a while (RPI3B)

masterchop, 2019-02-03:
I ran through the steps:

$ sudo apt-get install python-pip python3-pip python3-scipy libhdf5-dev
$ sudo apt-get install -y openmpi-bin libopenmpi-dev
$ sudo pip3 uninstall tensorflow
$ wget -O tensorflow-1.11.0-cp35-cp35m-linux_armv7l.whl https://github.com/PINTO0309/Tensorflow-bin/raw/master/tensorflow-1.11.0-cp35-cp35m-linux_armv7l_jemalloc.whl
$ sudo pip3 install tensorflow-1.11.0-cp35-cp35m-linux_armv7l.whl

And this has been the only package that has allow me to load a tflite model, thanks for that.
But i notice that is taking only 25% of the CPU is using 1 thread only making it slower than using regular tensorflow package with normal model.

so i tried to install:

tensorflow-1.11.0-cp35-cp35m-linux_armv7l_jemalloc.whl
tensorflow-1.11.0-cp35-cp35m-linux_armv7l_jemalloc_mpi.whl 
tensorflow-1.11.0-cp35-cp35m-linux_armv7l_jemalloc_mpi_multithread.whl 

but non of them work as i got: 
*.whl is not a supported wheel on this platform.

i am using a RTSP camera to get the images and script hangs after a while with no response i need to end the process manually something that doesnt happen with regular tensorflow.

am i doing something wrong? what am i missing i was expecting better results.

PINTO0309, 2019-02-03:
@masterchop 

>tensorflow-1.11.0-cp35-cp35m-linux_armv7l_jemalloc_mpi_multithread.whl

The installation method is as follows.
Just change the name of the architecture correctly.
```bash
$ mv tensorflow-1.11.0-cp35-cp35m-linux_armv7l_jemalloc.whl tensorflow-1.11.0-cp35-cp35m-linux_armv7l.whl
```
or
```
$ mv tensorflow-1.11.0-cp35-cp35m-linux_armv7l_jemalloc_mpi_multithread.whl tensorflow-1.11.0-cp35-cp35m-linux_armv7l.whl
```

And, Execute below.
```bash
$ sudo -H pip3 install tensorflow-1.11.0-cp35-cp35m-linux_armv7l.whl
```

However, the Python API does not seem to be multithread enabled.
**https://github.com/tensorflow/tensorflow/issues/21574#issuecomment-435536083**

masterchop, 2019-02-03:
Thanks for the reply, i got the same results 0.2 FPS but seems more stable.
By the way i see you recommend Tensorflow v1.12.0 rc0, Stand alone installer.
on that thread those it on that version?

PINTO0309, 2019-02-03:
@masterchop 
I will prepare a bit because the build script seems to be improved.
I will try it with the latest version so please give me some time.

PINTO0309, 2019-02-16:
Feb 16, 2019 Tensorflow "master" branch.
This is my memorandum.
I will verify from now.

**PythonAPI Multi Thread - During review**
**https://github.com/tensorflow/tensorflow/pull/25748**
**Avoiding build errors**
**https://github.com/tensorflow/tensorflow/issues/25120#issuecomment-464296755**

PINTO0309, 2019-02-16:
Feb 17, 2019 success.

**Build script**
**https://github.com/tensorflow/tensorflow/blob/master/tensorflow/lite/g3doc/rpi.md**

**Output destination path**
/home/pi/tensorflow/tensorflow/lite/tools/make/gen/rpi_armv7l/lib/libtensorflow-lite.a
/home/pi/tensorflow/tensorflow/lite/tools/make/gen/rpi_armv7l/bin/benchmark_model
/home/pi/tensorflow/tensorflow/lite/tools/make/gen/rpi_armv7l/bin/minimal

PINTO0309, 2019-02-23:
@masterchop 
Performance improved 2.5 times.
https://github.com/tensorflow/tensorflow/issues/21574#issuecomment-466615370

closed 2019-02-23 by PINTO0309

Xiadalei, 2019-05-24:
> 
> 
> Feb 16, 2019 Tensorflow "master" branch.
> This is my memorandum.
> I will verify from now.
> 
> **PythonAPI Multi Thread - During review**
> **[tensorflow/tensorflow#25748](https://github.com/tensorflow/tensorflow/pull/25748)**
> **Avoiding build errors**
> **[tensorflow/tensorflow#25120 (comment)](https://github.com/tensorflow/tensorflow/issues/25120#issuecomment-464296755)**
> 
> Feb 17, 2019 Success.
> 
> **Build script**
> **https://github.com/tensorflow/tensorflow/tree/master/tensorflow/lite/tools/pip_package**
> 
> **Output destination path**
> /tmp/tflite_pip/dist/tflite_runtime-1.12.0-cp27-cp27mu-linux_armv7l.whl
> 
> **Install script**
> 
> ```shell
> $ sudo -H pip install --upgrade /tmp/tflite_pip/dist/tflite_runtime-1.12.0-cp27-cp27mu-linux_armv7l.whl
> $ python
> >> import tflite_runtime as tflr
> ```
> 
> The following error occurs.
> 
> ```shell
> $ ldd -r _interpreter_wrapper.so 
> 	linux-vdso.so.1 (0x7eb3b000)
> 	/usr/lib/arm-linux-gnueabihf/libarmmem.so (0x76d2a000)
> 	libstdc++.so.6 => /usr/lib/arm-linux-gnueabihf/libstdc++.so.6 (0x76bc8000)
> 	libm.so.6 => /lib/arm-linux-gnueabihf/libm.so.6 (0x76b49000)
> 	libgcc_s.so.1 => /lib/arm-linux-gnueabihf/libgcc_s.so.1 (0x76b1c000)
> 	libpthread.so.0 => /lib/arm-linux-gnueabihf/libpthread.so.0 (0x76af3000)
> 	libc.so.6 => /lib/arm-linux-gnueabihf/libc.so.6 (0x769b4000)
> 	/lib/ld-linux-armhf.so.3 (0x76eed000)
> undefined symbol: PyExc_ValueError	(./_interpreter_wrapper.so)
> undefined symbol: _ZTVN6tflite19interpreter_wrapper19PythonErrorReporterE	(./_interpreter_wrapper.so)
> undefined symbol: PyExc_IndexError	(./_interpreter_wrapper.so)
> undefined symbol: PyExc_SyntaxError	(./_interpreter_wrapper.so)
> undefined symbol: _Py_NoneStruct	(./_interpreter_wrapper.so)
> undefined symbol: PyExc_AttributeError	(./_interpreter_wrapper.so)
> undefined symbol: PyExc_IOError	(./_interpreter_wrapper.so)
> undefined symbol: PyExc_MemoryError	(./_interpreter_wrapper.so)
> undefined symbol: PyExc_OverflowError	(./_interpreter_wrapper.so)
> undefined symbol: _PyWeakref_CallableProxyType	(./_interpreter_wrapper.so)
> undefined symbol: PyInstance_Type	(./_interpreter_wrapper.so)
> undefined symbol: PyExc_ZeroDivisionError	(./_interpreter_wrapper.so)
> undefined symbol: _tensorflow_numpy_api	(./_interpreter_wrapper.so)
> undefined symbol: PyExc_NotImplementedError	(./_interpreter_wrapper.so)
> undefined symbol: PyExc_RuntimeError	(./_interpreter_wrapper.so)
> undefined symbol: _Py_NotImplementedStruct	(./_interpreter_wrapper.so)
> undefined symbol: PyExc_TypeError	(./_interpreter_wrapper.so)
> undefined symbol: _PyWeakref_ProxyType	(./_interpreter_wrapper.so)
> undefined symbol: PyObject_GenericGetAttr	(./_interpreter_wrapper.so)
> undefined symbol: PyObject_GenericGetAttr	(./_interpreter_wrapper.so)
> undefined symbol: PyClass_Type	(./_interpreter_wrapper.so)
> undefined symbol: PyExc_SystemError	(./_interpreter_wrapper.so)
> undefined symbol: PyTuple_SetItem	(./_interpreter_wrapper.so)
> undefined symbol: PyLong_AsLong	(./_interpreter_wrapper.so)
> undefined symbol: _PyInstance_Lookup	(./_interpreter_wrapper.so)
> undefined symbol: PyObject_GetAttr	(./_interpreter_wrapper.so)
> undefined symbol: PyObject_Call	(./_interpreter_wrapper.so)
> undefined symbol: PyString_AsStringAndSize	(./_interpreter_wrapper.so)
> undefined symbol: PyObject_Init	(./_interpreter_wrapper.so)
> undefined symbol: PyObject_GetAttrString	(./_interpreter_wrapper.so)
> undefined symbol: PyErr_Restore	(./_interpreter_wrapper.so)
[... 28 lines cut ...]
> undefined symbol: PyCapsule_GetPointer	(./_interpreter_wrapper.so)
> undefined symbol: PyInt_AsLong	(./_interpreter_wrapper.so)
> undefined symbol: PyObject_Free	(./_interpreter_wrapper.so)
> undefined symbol: PyType_Ready	(./_interpreter_wrapper.so)
> undefined symbol: PyObject_Size	(./_interpreter_wrapper.so)
> undefined symbol: _ZN6tflite19interpreter_wrapper19PythonErrorReporter7messageB5cxx11Ev	(./_interpreter_wrapper.so)
> undefined symbol: _ZN6tflite19interpreter_wrapper19PythonErrorReporter9exceptionEv	(./_interpreter_wrapper.so)
> undefined symbol: PyLong_FromLong	(./_interpreter_wrapper.so)
> undefined symbol: PyErr_WriteUnraisable	(./_interpreter_wrapper.so)
> undefined symbol: PyObject_IsTrue	(./_interpreter_wrapper.so)
> undefined symbol: PyErr_Occurred	(./_interpreter_wrapper.so)
> undefined symbol: _ZN6tflite12python_utils27FillStringBufferWithPyArrayEP7_objectPNS_13DynamicBufferE	(./_interpreter_wrapper.so)
> undefined symbol: PyErr_Fetch	(./_interpreter_wrapper.so)
> undefined symbol: PyObject_CallFunctionObjArgs	(./_interpreter_wrapper.so)
> undefined symbol: PyArg_UnpackTuple	(./_interpreter_wrapper.so)
> undefined symbol: PyString_AsString	(./_interpreter_wrapper.so)
> undefined symbol: PyInt_FromLong	(./_interpreter_wrapper.so)
> undefined symbol: PyObject_Malloc	(./_interpreter_wrapper.so)
> undefined symbol: PyBool_FromLong	(./_interpreter_wrapper.so)
> ```
> 
> Next challenge.
> **https://github.com/tensorflow/tensorflow/blob/master/tensorflow/lite/g3doc/rpi.md**

so how do you fix the `undefined symbol: _ZTVN6tflite19interpreter_wrapper19PythonErrorReporterE` problem? (I encountered with that too)