PINTO0309/Tensorflow-bin #3: tensorflow-1.11.0-cp36-cp36m-linux_aarch64.whl failed.

bjtommychen, 2019-03-21:
Successfully installed tensorboard-1.11.0 tensorflow-1.11.0
[toybrick@localhost work]$ python3
Python 3.6.8 (default, Jan 31 2019, 09:06:51)
[GCC 8.2.1 20181215 (Red Hat 8.2.1-6)] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import tensorflow as tf
Traceback (most recent call last):
  File "/home/toybrick/.local/lib/python3.6/site-packages/tensorflow/python/pywrap_tensorflow.py", line 58, in <module>
    from tensorflow.python.pywrap_tensorflow_internal import *
  File "/home/toybrick/.local/lib/python3.6/site-packages/tensorflow/python/pywrap_tensorflow_internal.py", line 28, in <module>
    _pywrap_tensorflow_internal = swig_import_helper()
  File "/home/toybrick/.local/lib/python3.6/site-packages/tensorflow/python/pywrap_tensorflow_internal.py", line 24, in swig_import_helper
    _mod = imp.load_module('_pywrap_tensorflow_internal', fp, pathname, description)
  File "/usr/lib64/python3.6/imp.py", line 243, in load_module
    return load_dynamic(name, filename, file)
  File "/usr/lib64/python3.6/imp.py", line 343, in load_dynamic
    return _load(spec)
ImportError: libcublas.so.9.0: cannot open shared object file: No such file or directory

PINTO0309, 2019-03-21:
Make sure your environment is the same as the one below.

Python3.x + TX2 aarch64 (JetPack-L4T-3.3-linux-x64_b39)
- L4T R28.2.1（TX2 / TX2i）
- L4T R28.2（TX1）
- CUDA 9.0
- cuDNN 7.1.5
- TensorRT 4.0
- VisionWorks 1.6

bjtommychen, 2019-03-23:
sorry. I misunderstood it.

closed 2019-03-23 by bjtommychen