PINTO0309/Tensorflow-bin #4: ImportError: cannot import name 'cloud'

wyasuda, 2019-03-24:
Thank you for your great work.
I followed your instruction and installed tensorflow-2.0.0a0-cp35-cp35m-linux_armv7l.whl.
Then I run the command below.

python3 label_image.py --num_threads 4 --image grace_hopper.bmp --model_file mobilenet_v1_1.0_224_quant.tflite --label_file labels.txt

I got an error below.

Traceback (most recent call last):
  File "label_image.py", line 7, in <module>
    from tensorflow.contrib.lite.python import interpreter as interpreter_wrapper
  File "/usr/local/lib/python3.5/dist-packages/tensorflow/contrib/__init__.py", line 31, in <module>
    from tensorflow.contrib import cloud
ImportError: cannot import name 'cloud'

I use RaspberryPi Model B+and Raspbian 2018-11-13-raspbian-stretch image.

Your kind advice would be very helpful.

PINTO0309, 2019-03-24:
The module path of "lite" has been changed since Tensorflow v1.12.0.
Do not specify "contrib".

**label_image.py**
```python:label_image.py
from tensorflow.contrib.lite.python import interpreter as interpreter_wrapper
```
```python:label_image.py
from tensorflow.lite.python import interpreter as interpreter_wrapper
```

wyasuda, 2019-03-24:
It worked!
Thank you so much!

closed 2019-03-24 by wyasuda