PINTO0309/Tensorflow-bin #5: Error after installing tensorflow

RatheeshRK, 2019-05-11:
NVIDIA-Drive PX2
  
**[Required]** Details of the work you did before the problem occurred:  
<br> Installed tensorflow, pip list |grep tensorflow shows the correct version
<br> I had set the environment variables:LD_LIBRARY_PATH  as /usr/local/cuda-9.2/lib64
<br>
**[Required]** Error message:  
<br> ImportError: libcublas.so.9.0: cannot open shared object file: No such file or directory
<br>
<br>
**[Required]** Overview of problems and questions:  
<br>
<br>
<br>

PINTO0309, 2019-05-11:
As it is displayed in the error message.
The version of CUDA is different.
If you want to use CUDA version 9.2, please build Tensorflow yourself.

PINTO0309, 2019-07-27:
Close because there is no reply for 2 months or more.

closed 2019-07-27 by PINTO0309