PINTO0309/Tensorflow-bin #6: Configuration error

inagornyi, 2019-05-24:
pi@raspberrypi:~/tensorflow $ ./configure
WARNING: Running Bazel server needs to be killed, because the startup options are different.
WARNING: --batch mode is deprecated. Please instead explicitly shut down your Bazel server using the command "bazel shutdown".
You have bazel 0.19.2- (@non-git) installed.
Please upgrade your bazel installation to version 0.24.1 or higher to build TensorFlow!

PINTO0309, 2019-05-24:
Write about the version of Tensorflow you are about to compile. If it is 1.14.0-rc0, please build Bazel 0.24.1 yourself.

inagornyi, 2019-05-24:
tensorflow-1.13.1

inagornyi, 2019-05-24:
I tried to install via pip3 install tensflow.

>>> from tensorflow.contrib import predictor
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python3.5/dist-packages/tensorflow/contrib/__init__.py", line 40, in <module>
    from tensorflow.contrib import distribute
  File "/usr/local/lib/python3.5/dist-packages/tensorflow/contrib/distribute/__init__.py", line 33, in <module>
    from tensorflow.contrib.distribute.python.tpu_strategy import TPUStrategy
  File "/usr/local/lib/python3.5/dist-packages/tensorflow/contrib/distribute/python/tpu_strategy.py", line 27, in <module>
    from tensorflow.contrib.tpu.python.ops import tpu_ops
  File "/usr/local/lib/python3.5/dist-packages/tensorflow/contrib/tpu/__init__.py", line 69, in <module>
    from tensorflow.contrib.tpu.python.ops.tpu_ops import *
  File "/usr/local/lib/python3.5/dist-packages/tensorflow/contrib/tpu/python/ops/tpu_ops.py", line 39, in <module>
    resource_loader.get_path_to_datafile("_tpu_ops.so"))
  File "/usr/local/lib/python3.5/dist-packages/tensorflow/contrib/util/loader.py", line 56, in load_op_library
    ret = load_library.load_op_library(path)
  File "/usr/local/lib/python3.5/dist-packages/tensorflow/python/framework/load_library.py", line 61, in load_op_library
    lib_handle = py_tf.TF_LoadLibrary(library_filename)
tensorflow.python.framework.errors_impl.InvalidArgumentError: Invalid name: 
An op that loads optimization parameters into HBM for embedding. Must be
preceded by a ConfigureTPUEmbeddingHost op that sets up the correct
embedding table configuration. For example, this op is used to install
parameters that are loaded from a checkpoint before a training loop is
executed.

parameters: A tensor containing the initial embedding table parameters to use in embedding
lookups using the Adagrad optimization algorithm.
accumulators: A tensor containing the initial embedding table accumulators to use in embedding
lookups using the Adagrad optimization algorithm.
table_name: Name of this table; must match a name in the
  TPUEmbeddingConfiguration proto (overrides table_id).
num_shards: Number of shards into which the embedding tables are divided.
shard_id: Identifier of shard for this operation.
table_id: Index of this table in the EmbeddingLayerConfiguration proto
  (deprecated).

inagornyi, 2019-05-24:
Help me. I already try to setup tensorflow on raspberry pi a few days.

PINTO0309, 2019-05-24:
```
$ sudo apt-get install -y libhdf5-dev libc-ares-dev libeigen3-dev
$ sudo pip3 install keras_applications==1.0.7 --no-deps
$ sudo pip3 install keras_preprocessing==1.0.9 --no-deps
$ sudo pip3 install h5py==2.9.0
$ sudo apt-get install -y openmpi-bin libopenmpi-dev
$ sudo pip3 uninstall tensorflow
$ wget -O tensorflow-1.13.1-cp35-cp35m-linux_armv7l.whl https://github.com/PINTO0309/Tensorflow-bin/raw/master/tensorflow-1.13.1-cp35-cp35m-linux_armv7l.whl
$ sudo pip3 install tensorflow-1.13.1-cp35-cp35m-linux_armv7l.whl
```

```
$ python3
>> import tensorflow as tf
```

inagornyi, 2019-05-24:
```
>>> import tensorflow as tf
>>> from tensorflow.contrib import predictor
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python3.5/dist-packages/tensorflow/contrib/__init__.py", line 30, in <module>
    from tensorflow.contrib import cloud
ImportError: cannot import name 'cloud'
```

PINTO0309, 2019-05-24:
The installation of Tensorflow in your environment has been successfully completed.
However, there seems to be an essential bug in "contrib" of Tensorflow v1.13.1.
https://github.com/lhelontra/tensorflow-on-arm/issues/52

It would be better to use Tensorflow v1.12.0 or to abandon importing the contrib package.

PINTO0309, 2019-07-19:
@ghost
Please see below.
https://github.com/PINTO0309/Tensorflow-bin/issues/7#issuecomment-513190273

PINTO0309, 2019-07-27:
Fixed 7fc2cff00d2fc41eaecea15944f0c730773fc968

closed 2019-07-27 by PINTO0309