PINTO0309/Tensorflow-bin #7: Python 3.7 support on the Raspberry Pi?

aallan, 2019-06-11:
With Debian Buster coming close to a release, do you have a timeline for Python 3.7 prebuilt binaries for Tensorflow Lite?

PINTO0309, 2019-06-11:
I will try to build when the next version of Raspberry Pi's OS is officially released.

aallan, 2019-06-24:
…now you see why I was poking around for this, https://blog.hackster.io/benchmarking-machine-learning-on-the-new-raspberry-pi-4-model-b-88db9304ce4.

PINTO0309, 2019-06-24:
Ah! I already retweeted your tweets and shared them with Japanese engineers.
Is the image of 64bit Buster already released?
It is an old preview version, but this is the latest version.
https://wiki.debian.org/RaspberryPi3

aallan, 2019-06-24:
Possibly the best person to ask is @bennuttall. But Raspbian images on the download page now seem to be pointing to Raspbian Buster? https://www.raspberrypi.org/downloads/raspbian/

PINTO0309, 2019-06-24:
I created a 64bit Debian image. I do not know if it is good to install Python3.7 on this.
https://github.com/PINTO0309/RaspberryPi-bin

PINTO0309, 2019-06-24:
I succeeded in running 64bit Bazel on 64bit RaspberryPi3. Today, when I get home after work, I would like to try a native build of Tensorflow v1.14.0 aarch64. I will enable Tensorflow Lite and MultiThreading.

- Debian aarch64
- Python 3.7.3
- RaspberryPi3 model B+

PINTO0309, 2019-06-29:
Ubuntu16.04 x86_64 + QEMU Static mode + Debian Buster arm64(aarch64) + Python 3.7.3.
**https://qiita.com/PINTO/items/b089d9d5e4490d9c986c**

Tensorflow v1.14.0 build sequence (during creation).
**https://github.com/PINTO0309/Tensorflow-bin/wiki/64bit-debian**

jingw222, 2019-07-04:
Big shoutout to your excellent work for the community. Is there any plans on releasing builds TensorFlow 2.0 (Python 3.6+) for ARM, something like `tensorflow-2.0.0-cp36-none-linux_armv7l.whl` or `tensorflow-2.0.0-cp37-none-linux_armv7l.whl`.

PINTO0309, 2019-07-04:
I will start when RaspberryPi 4 starts selling in Japan. Since RaspberryPi3 has few resources such as RAM, it takes about 3 days to build. And now I am very busy and I do not have enough time to work. Please wait a little more.

bafu, 2019-07-05:
@aallan I can run **v1.14.0** on **RPi 4** with these manual modifications, and hope that this helps before @PINTO0309 shares the new builds with the community:

1. Rename Wheel name from cp35 to cp37

    ```
    $ mv tensorflow-1.14.0-cp35-cp35m-linux_armv7l.whl tensorflow-1.14.0-cp37-cp37m-linux_armv7l.whl
    ```

1. Comment out L30 and L31 in `/home/pi/.local/lib/python3.7/site-packages/tensorflow/contrib/__init__.py`

    ```
    #if os.name != "nt" and platform.machine() != "s390x":
    #  from tensorflow.contrib import cloud
    ```

1. Change `tf.contrib.lite` to `tf.lite` if there is any in your codes.

---
I can get inference time around 300 ~ 350 ms with MobileNet SSD, FYI.

aallan, 2019-07-05:
> I can get inference time around 300 ~ 350 ms with MobileNet SSD, FYI.

Huh. That's sort of odd, I'm [seeing ~270ms inferencing times](https://blog.hackster.io/benchmarking-machine-learning-on-the-new-raspberry-pi-4-model-b-88db9304ce4) with full blown TensorFlow. Which MobileNet model are you using for that? Obviously going to have to sit down and do some benchmarks.

aallan, 2019-07-05:
> Obviously going to have to sit down and do some benchmarks.

…and yup. I'm getting 82ms inferencing times for my set up, which is around the ×2 to ×3 speed up over TensorFlow I was expecting. But, still. Wow. That's seriously impressive. The new Pi4 just beat both the NVIDIA Jetson Nano and the Intel NCS2.

PINTO0309, 2019-07-06:
QEMU 4.0.0 + Debian Buster aarch64 -> Bazel 0.24.1, build succeeded.
**https://github.com/PINTO0309/Bazel_bin/tree/master/0.24.1**

Now I have started making Tensorflow v1.14.0 + Python 3.7.3.

Although it is in Japanese, the environment construction procedure was described in the following URL.
**https://qiita.com/PINTO/items/e117bb0389f2163e2ac8**

bafu, 2019-07-07:
@aallan Thanks for sharing the [informative benchmark articles](https://blog.hackster.io/benchmarking-tensorflow-lite-on-the-new-raspberry-pi-4-model-b-3fd859d05b98) and I really enjoy them!

The mainly difference between our tests is the models we used ([model source](https://github.com/tensorflow/models/blob/master/research/object_detection/g3doc/detection_model_zoo.md)):

||ssdlite_mobilenet_v2_coco|ssd_mobilenet_v2_coco|
|---|---|---|
|Running|First runtime: 378.74 ms<br>CPU%: 240.2<br>[image]|First runtime: 118.11 ms<br>CPU%: 367.9<br>[image]|
|Complete|Avg. runtime: **282.42 ms**[image]|Avg. runtime: **116.03 ms**[image]|

The interesting thing is that when using `ssdlite_mobilenet_v2_coco`, CPU usage seems not to reach 4-thread, even `set_num_threds(4)` is applied.

PINTO0309, 2019-07-07:
@bafu 
The multi-threaded implementation of Tensorflow Lite works only for the Convolution layer.
**https://github.com/tensorflow/tensorflow/issues/21574#issuecomment-467255015**

PINTO0309, 2019-07-08:
I first created and committed an installer for Debian buster aarch64 + Python 3.7.3. Tensorflow v1.14.0. Next I will generate an installer for Debian (Raspbian) Buster armhf.

PINTO0309, 2019-07-12:
@aallan  @bafu
I released Tensorflow v1.14.0 armhf + Python 3.7.3 + Buster.

bafu, 2019-07-13:
@PINTO0309 Thank you for your efforts and it's really helpful for speeding up the dev process!

PINTO0309, 2019-07-13:
@jingw222 
I released tensorflow-2.0.0b1-cp37-cp37m-linux_armv7l.whl + Python 3.7.3 + Raspbian Buster.

jingw222, 2019-07-14:
@PINTO0309 Thanks a lot. I really appreciate this.

PINTO0309, 2019-07-15:
I have released "Tensorflow v2.0.0-beta1 for Debian Buster aarch64, Tensorflow Lite / MultiThread enabled".

romilly, 2019-07-19:
I'm getting this error when running
`import tensorflow.contrib.tensorrt as trt`
 from tensorflow-1.14.0-cp37-cp37m-linux_armv7l.whl:

`ImportError: cannot import name 'cloud' from 'tensorflow.contrib' (/home/pi/.local/share/virtualenvs/bench-x9AggLjp/lib/python3.7/site-packages/tensorflow/contrib/__init__.py)
`
The workaround is to edit out the lines that try to import cloud:

`#if os.name != "nt" and platform.machine() != "s390x":`
`#  from tensorflow.contrib import cloud`

but ISTR this is an old bug that can be avoided by a compile-time option.

PINTO0309, 2019-07-19:
@romilly 
Thank you for pointing out! I appreciate it very much. I will try to rebuild and re-release the package when I have time.

romilly, 2019-07-21:
@PINTO0309 Thank you! Your skill and  hard work are very much appreciated.

bafu, 2019-07-22:
@romilly For the `s390x` issue, you can check your source code to see if there is any `tf.contrib.lite`, and replace all of them by `tf.lite`.

romilly, 2019-07-22:
I have. There isn't.

romilly, 2019-07-22:
This appears to be a recurrence of
https://github.com/tensorflow/tensorflow/issues/23976

PINTO0309, 2019-07-22:
@romilly , @bafu 
A bug fix seems to have been merged into the master branch a few days ago.
https://github.com/tensorflow/tensorflow/commit/ed62ac8203999513dfae03498e871ea35eb60cc4

PINTO0309, 2019-07-26:
@romilly , @bafu
I implemented the solution for the problem and started rebuilding. (https://github.com/tensorflow/tensorflow/issues/23976)
QEMU 4.0.0 + Debian Buster armhf + Tensorflow v1.14.0

**How to create a Debian Buster armhf OS image from scratch in hardware emulation mode of QEMU 4.0.0 (Kernel 4.19.0-5-armmp-lpae, for building Tensorflow armhf)**
**https://qiita.com/PINTO/items/c10283a28d0699f01e01**

romilly, 2019-07-26:
> tensorflow/tensorflow#23976

Thanks again for all your hard work.

PINTO0309, 2019-07-27:
@romilly, @bafu  
#29617 Fixed all Wheel files.

**tensorflow/contrib/\_\_init\_\_.py**
```python
from tensorflow.contrib import checkpoint
#if os.name != "nt" and platform.machine() != "s390x":
#  from tensorflow.contrib import cloud
from tensorflow.contrib import cluster_resolver
```
```python
from tensorflow.contrib.summary import summary

if os.name != "nt" and platform.machine() != "s390x":
  try:
    from tensorflow.contrib import cloud
  except ImportError:
    pass

from tensorflow.python.util.lazy_loader import LazyLoader
ffmpeg = LazyLoader("ffmpeg", globals(),
                    "tensorflow.contrib.ffmpeg")
```

PINTO0309, 2019-08-31:
You're welcome. I have released Tensorflow v2.0.0-rc0 for Raspbian Buster armhf.

PhilipXue, 2019-09-04:
@bafu 
I think I found the reason for your counter-intuitive result. The model of mobilenet v2 ssd @aallan used is a quantized model which using int8 for reference. This is much faster than its fp32 counterpart (but will sacrifice some performance in precision).
The speed for a fp32 mobilenet v2 ssd should 400+ ms/img according to my test.

PINTO0309, 2019-09-04:
I created a standalone installer for Tensorflow Lite v1.14.0 / v2.0.0-rc0. The installer size is limited to 1.6MB. I will check the operation at a later date.
**https://github.com/PINTO0309/TensorflowLite-bin.git**

PINTO0309, 2020-07-13:
Closed as already resolved.

closed 2020-07-13 by PINTO0309