PINTO0309/Tensorflow-bin #9: Question on performance

previ, 2019-07-10:
Device: RaspberryPi3+:  
  
CPU architecture: armv7l
  
OS: Raspbian stretch
 
Conditions
I installed tensor flow lite for python 3.5, multi threaded (this wheel: tensorflow-1.14.0-cp35-cp35m-linux_armv7l.whl).
I run the test as documented in the readme.

Question
I cannot get better timing than 0.27 seconds with num_threads=4 while the readme reports a much faster 0.16.
Any hints?
Thanks!

PINTO0309, 2019-07-10:
Please try **tensorflow-1.13.1-cp35-cp35m-linux_armv7l.whl** or **tensorflow-1.11.0-cp35-cp35m-linux_armv7l.whl**. Tensorflow is bloated as the version goes up. I have not yet clearly compared the performance.

previ, 2019-07-12:
I did some more tests, I could not get any better timing until, guess what? I switched the power input from my PC USB output to a proper 2.1A power supply.
Now I get timings consistent with your test, actually as short as:
0.127 seconds with ts 1.14
0.371 with ts 1.13.1
0.317 with ts 1.11

So 1.14 looks much faster than predecessors, I did not try 2.0 beta.

Thanks

PINTO0309, 2019-07-13:
@previ 
It is a very interesting result. By the way, I am using a 5V 3.0A high power AC adapter. It is only speculation, but the following may have an impact.

1. Power supply capacity of AC adapter
2. Difference in weight loading time between the first and second inference processes

PINTO0309, 2019-09-11:
@JCStarry 
I encourage you to try the following. It may be a little faster.

**Tensorflow Lite stand alone Installer**
**https://github.com/PINTO0309/TensorflowLite-bin.git**

**MobileNetV2**
**https://storage.googleapis.com/download.tensorflow.org/models/tflite_11_05_08/mobilenet_v2_1.0_224_quant.tgz**

**https://github.com/tensorflow/tensorflow/issues/31229#issuecomment-527296093**

JCStarry, 2019-09-16:
> @JCStarry
> I encourage you to try the following. It may be a little faster.
> 
> **Tensorflow Lite stand alone Installer**
> **https://github.com/PINTO0309/TensorflowLite-bin.git**
> 
> **MobileNetV2**
> **https://storage.googleapis.com/download.tensorflow.org/models/tflite_11_05_08/mobilenet_v2_1.0_224_quant.tgz**
> 
> **[tensorflow/tensorflow#31229 (comment)](https://github.com/tensorflow/tensorflow/issues/31229#issuecomment-527296093)**

[image]
This is the result of testing MobileNetV2 ，ありがとう！！for  your work

PINTO0309, 2020-07-13:
As there has been no progress, I'm closing it.

closed 2020-07-13 by PINTO0309