PINTO0309/Tensorflow-bin #11: THANK YOU!!!

Tylersuard, 2019-07-28:
**[Required]** Your device (RaspberryPi3, LaptopPC, or other device name):  
  Pi Zero W

**[Required]** Your device's CPU architecture (armv7l, x86_64, or other architecture name):  
  ARMV6?

**[Required]** Your OS (Raspbian, Ubuntu1604, or other os name):  
Buster  

**[Required]** Details of the work you did before the problem occurred:  
I looked around the internet for 2 hours trying to figure out how to get TFlite onto a pi.  Every tutorial involved complicated, time-consuming cross-compilation.  Than you for making it easier.
**[Required]** Error message:  
None
**[Required]** Overview of problems and questions:  
It looks like you already did the job of cross-compiling.  If you don't mind, in beginner-speak, could you explain how to use and install these things onto a Pi4b or a Pi Zero?  Thank you!

PINTO0309, 2019-07-28:
@Tylersuard 

**For RaspberryPi4 model B**
```bash
$ wget https://github.com/PINTO0309/Tensorflow-bin/raw/master/tensorflow-1.14.0-cp37-cp37m-linux_armv7l.whl
$ sudo pip3 uninstall tensorflow
$ sudo -H pip3 instrall tensorflow-1.14.0-cp37-cp37m-linux_armv7l.whl
```

**For RaspberryPi Zero W**
Unfortunately, for now I only offer armv7l and aarch64 wheel files. It must be built with the armv6 architecture.

Tylersuard, 2019-07-28:
Ok well this is close enough, it will get me going with the pi 4.  Again,
thank you!!

PINTO0309, 2020-07-13:
As there has been no progress, I'm closing it.

closed 2020-07-13 by PINTO0309