PINTO0309/Tensorflow-bin #14: ARMv6 support missing

baqwas, 2019-09-02:
The installation steps for using **tf** under _Buster_, as outlined in the documentation, do not work. Any plans to extend this invaluable endeavor to the following?

- ARMv6-compatible processor rev 7 (v6l)
- Linux 4.19.66+ armv6l GNU/Linux (Buster)

Kind regards.

PINTO0309, 2019-09-02:
@baqwas 
Interesting. However, for verification I must first purchase RaspberryPi zero or other armv6l hardware. Is the device you are using RaspberryPi Zero?

baqwas, 2019-09-02:
@PINTO0309 

I'm in the U.S. I will gladly send you the current RPi0 W v1.3 anywhere in the world (assuming that you will take care of any "customs" fees). Not asking for any schedule commitments on porting to the RPi0 platform - just use your experience and provide feedback on feasibility. Of course, if you are successful in porting then I would like to enjoy in testing the solution. Thanks for replying.

Kind regards.

closed 2019-09-02 by baqwas