PINTO0309/Tensorflow-bin #17: 'GLIBC_2.29' error not found for Python 3.7 + Tensorflow v2

shineofleo, 2019-10-02:
Thank you and I was going to install Python 3.7 + Tensorflow v2 on my raspberry pi 4, Buster.

Everything went well and pip3 list can see the v2 tensorflow. But when import the tf in python3, below error happened:

Traceback (most recent call last):
  File "/home/pi/.local/lib/python3.7/site-packages/tensorflow_core/python/pywrap_tensorflow.py", line 58, in <module>
    from tensorflow.python.pywrap_tensorflow_internal import *
  File "/home/pi/.local/lib/python3.7/site-packages/tensorflow_core/python/pywrap_tensorflow_internal.py", line 28, in <module>
    _pywrap_tensorflow_internal = swig_import_helper()
  File "/home/pi/.local/lib/python3.7/site-packages/tensorflow_core/python/pywrap_tensorflow_internal.py", line 24, in swig_import_helper
    _mod = imp.load_module('_pywrap_tensorflow_internal', fp, pathname, description)
  File "/usr/lib/python3.7/imp.py", line 242, in load_module
    return load_dynamic(name, filename, file)
  File "/usr/lib/python3.7/imp.py", line 342, in load_dynamic
    return _load(spec)
ImportError: /lib/arm-linux-gnueabihf/libm.so.6: version 'GLIBC_2.29' not found (required by /home/pi/.local/lib/python3.7/site-packages/tensorflow_core/python/_pywrap_tensorflow_internal.so)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/pi/.local/lib/python3.7/site-packages/tensorflow/__init__.py", line 98, in <module>
    from tensorflow_core import *
  File "/home/pi/.local/lib/python3.7/site-packages/tensorflow_core/__init__.py", line 40, in <module>
    from tensorflow.python.tools import module_util as _module_util
  File "/home/pi/.local/lib/python3.7/site-packages/tensorflow/__init__.py", line 50, in __getattr__
    module = self._load()
  File "/home/pi/.local/lib/python3.7/site-packages/tensorflow/__init__.py", line 44, in _load
    module = _importlib.import_module(self.__name__)
  File "/usr/lib/python3.7/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "/home/pi/.local/lib/python3.7/site-packages/tensorflow_core/python/__init__.py", line 49, in <module>
    from tensorflow.python import pywrap_tensorflow
  File "/home/pi/.local/lib/python3.7/site-packages/tensorflow_core/python/pywrap_tensorflow.py", line 74, in <module>
    raise ImportError(msg)
ImportError: Traceback (most recent call last):
  File "/home/pi/.local/lib/python3.7/site-packages/tensorflow_core/python/pywrap_tensorflow.py", line 58, in <module>
    from tensorflow.python.pywrap_tensorflow_internal import *
  File "/home/pi/.local/lib/python3.7/site-packages/tensorflow_core/python/pywrap_tensorflow_internal.py", line 28, in <module>
    _pywrap_tensorflow_internal = swig_import_helper()
  File "/home/pi/.local/lib/python3.7/site-packages/tensorflow_core/python/pywrap_tensorflow_internal.py", line 24, in swig_import_helper
    _mod = imp.load_module('_pywrap_tensorflow_internal', fp, pathname, description)
  File "/usr/lib/python3.7/imp.py", line 242, in load_module
    return load_dynamic(name, filename, file)
  File "/usr/lib/python3.7/imp.py", line 342, in load_dynamic
    return _load(spec)
ImportError: /lib/arm-linux-gnueabihf/libm.so.6: version `GLIBC_2.29' not found (required by /home/pi/.local/lib/python3.7/site-packages/tensorflow_core/python/_pywrap_tensorflow_internal.so)

Failed to load the native TensorFlow runtime.

See https://www.tensorflow.org/install/errors

for some common reasons and solutions.  Include the entire stack trace
above this error message when asking for help.

I tried the 【Appendix】 C Library + Tensorflow v1.x.x / v2.x.x you provided but not helping.

Any good idea how to go on?

PINTO0309, 2019-10-02:
Same issue. https://github.com/PINTO0309/Tensorflow-bin/issues/10
I am in the process of building the latest Tensorflow v2, so the build in the environment you want is likely to be far ahead. If you are in a hurry, I suggest you build it yourself.

PINTO0309, 2019-10-02:
I'm building with the latest OS image that was updated on September 26, 2019.
**https://www.raspberrypi.org/downloads/raspbian/**
[image]
[image]
[image]

shineofleo, 2019-10-02:
Thanks for the super fast reply.

I am on the same 10 (buster) as the rsult of lsb_release, but the ldd --version output shows glibc = 2.28.

So far the buster only provides 2.28 as I can check. I am tempted to compile and upgrade the glibc to 2.29 by myself, but not sure if it will break everything. Actually, glibc is updated with 2.30 now!

filipvg68, 2019-10-18:
@shineofleo : how did you upgrade to glibc 2.30?

PINTO0309, 2019-10-18:
Simple installation procedure of glibc2.29. glibc2.29 is installed at the same time as openjdk-8-jdk is installed. It is recommended to take an image backup because the environment may become unstable.
**[How to install openjdk-8-jdk on Raspbian Buster armhf](https://qiita.com/PINTO/items/a6ae8e04d382493ef369)**
or
**[How to install openjdk-8-jdk on Debian Buster (Debian 10) armhf](https://qiita.com/PINTO/items/5445c5e899f68d928f0d)**

PINTO0309, 2019-10-20:
@shineofleo 
The glibc2.28 version is currently being rebuilt. If the build is successful, it will be released tomorrow.

PINTO0309, 2019-10-22:
@shineofleo
Committed TF v2.0.0 wheel for glibc2.28.

closed 2019-10-22 by PINTO0309

sgabor1, 2020-01-14:
@PINTO0309 
Same error happened to me after installing following package:
> tensorflow-2.1.0-cp37-cp37m-linux_armv7l.whl | ○ | Raspbian/Debian Buster, glibc 2.28
TF 2.1 supposed to be using glibc 2.28 as per the readme..?

PINTO0309, 2020-01-14:
The official Tensorflow site does not specify a glibc version. The glibc version depends on the environment where .whl is built. I always build natively using the official OS images for RaspberryPi and Ubuntu, so it depends on the glibc version that comes with the latest OS at the moment. Both Debian Buster / Raspbian Buster are glibc 2.28. To run on an OS with a version other than glibc 2.28, you need to build natively or cross compile by yourself.

sgabor1, 2020-01-14:
I have glibc 2.28 but get this error when trying to import tensorflow after installing the latest tensorflow-2.1.0-cp37-cp37m-linux_armv7l.whl package:
import tensorflow as tf
`ImportError: /lib/arm-linux-gnueabihf/libm.so.6: version 'GLIBC_2.29' not found (required by /home/pi/virtualenvs/env/lib/python3.7/site-packages/tensorflow_core/python/_pywrap_tensorflow_internal.so)`

ldd --version
`ldd (Debian GLIBC 2.28-10+rpi1) 2.28`

PINTO0309, 2020-01-14:
I may have inadvertently upgraded glibc...
Isn't the problem solved by executing the following command?

```bash
$ sudo apt-get update
$ sudo apt-get upgrade
```

PINTO0309, 2020-01-14:
@sgabor1 
It looks like I made a mistake. I initialize the SD card and rebuild Tensorflow from scratch. Please wait a day if necessary.

PINTO0309, 2020-01-15:
@sgabor1 
I rebuilt and committed with glibc 2.28.

sgabor1, 2020-01-15:
@PINTO0309 
It's working fine now, thank you so much!!

zanazakaryaie, 2020-03-22:
Hi
I'm on Raspberry Pi 3B+ with Ubuntu 18.04 Mate (aarch64). Ubuntu Mate 18.04 comes with Python 3.6. I installed Python 3.7 and downloaded `tensorflow-1.15.0-cp37-cp37m-linux_aarch64.whl`. After installation, I get this error:

> version `GLIBC_2.28' not found

`led --version` prints `2.27`. How should I upgrade to `GLIBC 2.28`?

PINTO0309, 2020-03-22:
@zanazakaryaie 
Upgrading the compiler version is very difficult and is not recommended as it makes the whole system unstable. So I present the solutions in the recommended order.
1. Natively build Tensorflow on Ubuntu18.04 by yourself. Since Pi3 has very few resources, it takes 3 days to complete the build. 
 **https://github.com/PINTO0309/Tensorflow-bin#build-parameter**
2. You install Ubuntu19.10 aarch64 (Python3.7 is installed by default). It will end in an hour.
3. Wait for me to set up a build environment for Ubuntu18.04 and build natively (I'm busy at the end of the year so it will take a week or two)

zanazakaryaie, 2020-03-23:
@PINTO0309 
Thanks for the clear explanations. I will install Ubuntu 19.10