PINTO0309/Tensorflow-bin #18: tensorflow-2.0.0-cp37-cp37m-linux_aarch64.whl, which os image should we use?

ghimiredhikura, 2019-10-29:
Hi, 

Thank you for the wonderful work. I would like to run some benchmarking on with aarch64 version in both RPi3B+ and RPi4B. 

Which OS Image are you using for the corresponding one. Some info will be greatly appreciated. 

Thanks. 

Best,
Deepak

PINTO0309, 2019-10-29:
My aarch64 installer is intended for use in a 64-bit Debian environment other than RaspberryPi. Unfortunately, Raspbian 64bit image for RaspberryPi4 has not been released yet. However, a preview of the Raspbian 64bit image for RaspberryPi3 is available.

**RaspberryPi3 Preview Image**
**https://people.debian.org/~gwolf/raspberrypi3/20190628/**

If you need a 64-bit Pi4 image, you will need to build it yourself using Yocto etc.

**Pi4 64bit kernel build**
**https://www.raspberrypi.org/forums/viewtopic.php?t=244792**

ghimiredhikura, 2019-10-29:
Hi @PINTO0309, 

Thanks for the quick response. So "tensorflow-2.0.0-cp37-cp37m-linux_aarch64.whl" will work in both Pi3 and Pi4?

PINTO0309, 2019-10-29:
>So "tensorflow-2.0.0-cp37-cp37m-linux_aarch64.whl" will work in both Pi3 and Pi4?

It should work if you have 64-bit Debian Buster (aarch64) installed.

>One more help! RaspberryPi3 Preview Image will not work for Pi3B+ right?

I have confirmed that it works with Pi3 B+.

closed 2019-11-02 by PINTO0309

PINTO0309, 2019-11-02:
64-bit image creation repository
or **https://github.com/drtyhlpr/rpi23-gen-image.git**