PINTO0309/Tensorflow-bin #19: Can't install 2.1.0 on Python 3.7 Raspberry Pi 4

gilescoope, 2020-01-29:
It gets stuck on the "Building wheels for scipy" stage.
Firstly if the fortran compiler isn't found it fails. But even after I run
sudo apt-get install gfortran
I get errors

PINTO0309, 2020-01-30:
If possible, post the full text of all executed commands and error messages. I can't grasp the situation.

gilescoope, 2020-01-30:
I've got this working now. The problem is that scipy 1.4.1, a requirement for Tensorflow 2.1, doesn't have a prebuilt wheel so has to be built from source.

This has some dependencies, I got it to install using
sudo apt-get install gcc gfortran python-dev libgfortran5 libatlas3-base libatlas-base-dev libopenblas-dev libopenblas-base libblas-dev liblapack-dev cython 
sudo pip3 install pybind11

PINTO0309, 2020-02-08:
@gilescoope 
Thank you for pointing that out. Closed because it was reflected in the README.

closed 2020-02-08 by PINTO0309