PINTO0309/Tensorflow-bin #21: Do i need 3.7.3 ?

hugokoopmans, 2020-05-02:
**[Required]** Your device (RaspberryPi3, LaptopPC, or other device name):  
RaspberryPi4
  
**[Required]** Your device's CPU architecture (armv7l, x86_64, or other architecture name):  
  armv7

**[Required]** Your OS (Raspbian, Ubuntu1604, or other os name):  
Rasbian Buster
  
**[Required]** Details of the work you did before the problem occurred:  
Expected wheel to install
  
**[Required]** Error message:  
(serval) pi@serval-001:~ $ pip3 install tensorflow-1.15.0-cp37-cp37m-linux_armv7l.whl
ERROR: tensorflow-1.15.0-cp37-cp37m-linux_armv7l.whl is not a supported wheel on this platform.
(serval) pi@serval-001:~ $ python3
Python 3.7.7 (default, Apr  1 2020, 10:55:29) 
[GCC 9.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
  
**[Required]** Overview of problems and questions:  
On buster current version of python3 is 3.8.x
I installed python3.7 side by side and setup a virtual env with python3.7.**7** 

Do I specifically need **3.7.3**  for this wheel to install or do I miss something else here?

PINTO0309, 2020-05-02:
What are all the results of the commands below?
```
$ ldd --version
$ uname -m
$ uname -a
```

hugokoopmans, 2020-05-03:
pi@serval-001:~ $ ldd --version
ldd (Debian GLIBC 2.30-4) 2.30
Copyright (C) 2019 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
Written by Roland McGrath and Ulrich Drepper.
pi@serval-001:~ $ uname -m
armv7l
pi@serval-001:~ $ uname -a
Linux serval-001 4.19.97-v7l+ #1294 SMP Thu Jan 30 13:21:14 GMT 2020 armv7l GNU/Linux

PINTO0309, 2020-05-03:
Isn't your operating system **`Debian Bullseye`**, not **` Raspbian Buster`**?
**https://packages.debian.org/search?lang=en&suite=bullseye&searchon=names&keywords=glibc**

My installer doesn't support **`Bullseye`** yet.

[image]
===================================================
[image]

hugokoopmans, 2020-05-03:
hmmm intersting, I used **Raspbian Buster Lite** from this link https://downloads.raspberrypi.org/raspbian_lite_latest

hugokoopmans, 2020-05-03:
Can you point me to an image of Buster I could use? I will use that for my work then, sort of depend on this heel to install...

PINTO0309, 2020-05-03:
It's a bit old, but it is an image of Raspbian Buster released on September 26, 2019.

**Raspbian Buster Lite armv7l**
**https://drive.google.com/open?id=1Zx4hLuXm-1gB6-SZzMgZSF8L5F_amP8I**

Please update to the latest kernel with the following command after installation.
```bash
$ sudo apt update
$ sudo apt upgrade
```

hugokoopmans, 2020-05-03:
ok I have found my issue, was a problem on my site , long story but appeared I also added de debian sid repo in one of my custom install scripts, due to manual error this was an old part of my code and should have been removed...

closed 2020-05-03 by hugokoopmans