PINTO0309/Tensorflow-bin #22: GLIBC_2.28' not found

ArefAz, 2020-05-21:
device name:  RaspberryPi 4
  
architecture name: aarch64
  
OS:  Ubuntu 18.04

Python version: 3.7.5
  
Details of the work you did before the problem occurred:  I tried different versions of tensorflow that you provided (very much appreciated). When I try to import tensorflow the error shows up. Also, the tflite_runtime module is not installed this way, so I tried installing it from [here](https://www.tensorflow.org/lite/guide/python). But the problem is with the set_num_threads method. In 2.x tensorflow Interpreter has no module for this and because of this I cannot have the full utilization of the cpu.
  
Error message: 

>  ImportError: /lib/aarch64-linux-gnu/libc.so.6: version `GLIBC_2.28' not found (required by /usr/local/lib/python3.7/dist-packages/tensorflow_core/python/_pywrap_tensorflow_internal.so)

> AttributeError: 'Interpreter' object has no attribute 'set_num_threads'

  My GLIBC version is 2.27 from `ldd --version`. I searched about upgrading it but I found  it is risky and it may corrupt my other binary files.

Note: I reached 10fps on raspberry pi3 on ubuntu 19.04 but the same image of it would not boot on raspberry pi4 so I tried ubuntu 18.04 on raspberry 4 instead but I now have this problem and could only get less than 6 fps. I also could reach 10fps on this raspberry 4 using python 2.7.

PINTO0309, 2020-05-22:
**Ubuntu 18.04 aarch64/armv7l**
**https://github.com/PINTO0309/TensorflowLite-bin/tree/master/2.2.0**
You can generate it in 10 minutes if you follow the steps I wrote in the README.

>I'm also curious about why Tensorflow has removed the set_num_threads method in version 2.x. Do you have idea about it?

Multi-Thread was not implemented in the first place.  However, it will be merged soon.
**https://github.com/tensorflow/tensorflow/pull/25748**

ArefAz, 2020-05-24:
I used [this wheel ](https://github.com/PINTO0309/TensorflowLite-bin/blob/master/1.15.0-rc3/tflite_runtime-1.15.0rc3-cp37-cp37m-linux_aarch64.whl) and the problem was solved.

Thanks again.

closed 2020-05-24 by ArefAz