PINTO0309/Tensorflow-bin #24: tf 1.15.0 and tensor2tensor on raspberry pi -- get_data_files problem

radiodee1, 2020-05-26:
**[Required]** Your device (RaspberryPi3, LaptopPC, or other device name):  raspberry pi
  
**[Required]** Your device's CPU architecture (armv7l, x86_64, or other architecture name):  armv7l
  
**[Required]** Your OS (Raspbian, Ubuntu1604, or other os name):  raspbian
  
**[Required]** Details of the work you did before the problem occurred:  designed a t2t model
  
**[Required]** Error message:  see below
  
**[Required]** Overview of problems and questions:  
  
I am seeing this error. I'm using your tf 1.15.0 for armhf and python3.7. Ultimately I will be using raspbian and buster. I also use tensor2tensor and this is where I see my error now.

File "/usr/local/lib/python3.7/dist-packages/tensor2tensor/bin/t2t_decoder.py", line 209, in main
    decode(estimator, hp, decode_hp)
  File "/usr/local/lib/python3.7/dist-packages/tensor2tensor/bin/t2t_decoder.py", line 110, in decode
    checkpoint_path=FLAGS.checkpoint_path)
  File "/usr/local/lib/python3.7/dist-packages/tensor2tensor/utils/decoding.py", line 227, in decode_from_dataset
    checkpoint_path=checkpoint_path)
  File "/usr/local/lib/python3.7/dist-packages/tensor2tensor/utils/decoding.py", line 316, in decode_once
    for num_predictions, prediction in enumerate(predictions):
  File "/usr/local/lib/python3.7/dist-packages/tensorflow_estimator/python/estimator/estimator.py", line 620, in predict
    input_fn, ModeKeys.PREDICT)
  File "/usr/local/lib/python3.7/dist-packages/tensorflow_estimator/python/estimator/estimator.py", line 996, in _get_features_from_input_fn
    result = self._call_input_fn(input_fn, mode)
  File "/usr/local/lib/python3.7/dist-packages/tensorflow_estimator/python/estimator/estimator.py", line 1116, in _call_input_fn
    return input_fn(**kwargs)
  File "/usr/local/lib/python3.7/dist-packages/tensor2tensor/data_generators/problem.py", line 803, in estimator_input_fn
    dataset_kwargs=dataset_kwargs)
  File "/usr/local/lib/python3.7/dist-packages/tensor2tensor/data_generators/problem.py", line 888, in input_fn
    self.dataset(**dataset_kwargs),
  File "/usr/local/lib/python3.7/dist-packages/tensor2tensor/data_generators/problem.py", line 653, in dataset
    contrib.slim().parallel_reader.get_data_files(data_filepattern))
  File "/usr/local/lib/python3.7/dist-packages/tensorflow_core/contrib/slim/python/slim/data/parallel_reader.py", line 316, in get_data_files
    raise ValueError('No data files found in %s' % (data_sources,))
ValueError: No data files found in /app/transformer/../data/t2t_data/chat_movie_30/chat_line_problem-dev*

The error says that the 'dev' files don't exist but in fact they do. I don't know how to fix this. In the printout above the error shows up in the test phase. I can run the t2t code in the train phase and the code will save the data files but later when it comes time to read them it crashes again.

PINTO0309, 2020-05-26:
This is an issue that has nothing to do with the installer of this repository, just ask StackOverflow. Maybe the executing user doesn't have access to the file.

radiodee1, 2020-05-27:
You may be right but I think if this were the underlying problem the error message would be 'Permission Denied', not 'ValueError: No data files found'. Do you have any other clues?

radiodee1, 2020-05-29:
Close this if you want. I have not found a solution but I take your point. Thank you for your time.

closed 2020-05-29 by radiodee1