PINTO0309/Tensorflow-bin #25: Permission denied

hdihd9162, 2020-07-07:
**[Required]** Your device (RaspberryPi3, LaptopPC, or other device name):  RaspberryPi3
  
**[Required]** Your device's CPU architecture (armv7l, x86_64, or other architecture name):  armv7l
  
**[Required]** Your OS (Raspbian, Ubuntu1604, or other os name):  Raspbian
  
**[Required]** Details of the work you did before the problem occurred:  install tensorflow
  
**[Required]** Error message:  bash: ./tensorflow-2.2.0-cp37-cp37m-linux_armv7l_download.sh: Permission denied
  
**[Required]** Overview of problems and questions:  its not working

PINTO0309, 2020-07-07:
Please try below.
```bash
$ sudo bash ./tensorflow-2.2.0-cp37-cp37m-linux_armv7l_download.sh
```

hdihd9162, 2020-07-07:
after that what i do?

PINTO0309, 2020-07-07:
**https://github.com/PINTO0309/Tensorflow-bin#usage**

hdihd9162, 2020-07-07:
after sudo -H pip3 install tensorflow-2.2.0-cp37-cp37m-linux_armv7l.whl
show this eror
THESE PACKAGES DO NOT MATCH THE HASHES FROM THE REQUIREMENTS FILE. If you have updated the package versions, please update the hashes. Otherwise, examine the package contents carefully; someone may have tampered with them.
    scipy==1.4.1; python_version >= "3" from https://www.piwheels.org/simple/scipy/scipy-1.4.1-cp37-cp37m-linux_armv7l.whl#sha256=ef4e1b837ece171cb99a957a68d2320e7e3c649e0008ed4efb62851f2ff45bf0 (from tensorflow==2.2.0):
        Expected sha256 ef4e1b837ece171cb99a957a68d2320e7e3c649e0008ed4efb62851f2ff45bf0
             Got        d59b2553ec90945508401dc9b28c5d7c9819d96ad7857dc885ebe688d77b1505

hdihd9162, 2020-07-07:
i try many time and after while
 sudo -H pip3 install tensorflow-2.2.0-cp37-cp37m-linux_armv7l.whl
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already satisfied: tensorflow==2.2.0 from file:///home/pi/Desktop/tensorflow/tensorflow-2.2.0-cp37-cp37m-linux_armv7l.whl in /usr/local/lib/python3.7/dist-packages (2.2.0)
Requirement already satisfied: astunparse==1.6.3 in /usr/local/lib/python3.7/dist-packages (from tensorflow==2.2.0) (1.6.3)
Requirement already satisfied: wrapt>=1.11.1 in /usr/local/lib/python3.7/dist-packages (from tensorflow==2.2.0) (1.12.1)
Requirement already satisfied: numpy<2.0,>=1.16.0 in /usr/lib/python3/dist-packages (from tensorflow==2.2.0) (1.16.2)
Requirement already satisfied: protobuf>=3.8.0 in /usr/local/lib/python3.7/dist-packages (from tensorflow==2.2.0) (3.12.2)
Requirement already satisfied: wheel>=0.26; python_version >= "3" in /root/.local/lib/python3.7/site-packages (from tensorflow==2.2.0) (0.34.2)
Requirement already satisfied: opt-einsum>=2.3.2 in /usr/local/lib/python3.7/dist-packages (from tensorflow==2.2.0) (3.2.1)
Requirement already satisfied: termcolor>=1.1.0 in /usr/local/lib/python3.7/dist-packages (from tensorflow==2.2.0) (1.1.0)
Requirement already satisfied: six>=1.12.0 in /root/.local/lib/python3.7/site-packages (from tensorflow==2.2.0) (1.15.0)
Requirement already satisfied: keras-preprocessing>=1.1.0 in /usr/local/lib/python3.7/dist-packages (from tensorflow==2.2.0) (1.1.0)
Requirement already satisfied: absl-py>=0.7.0 in /usr/local/lib/python3.7/dist-packages (from tensorflow==2.2.0) (0.9.0)
Requirement already satisfied: google-pasta>=0.1.8 in /usr/local/lib/python3.7/dist-packages (from tensorflow==2.2.0) (0.2.0)
Requirement already satisfied: tensorboard<2.3.0,>=2.2.0 in /usr/local/lib/python3.7/dist-packages (from tensorflow==2.2.0) (2.2.2)
Requirement already satisfied: grpcio>=1.8.6 in /usr/local/lib/python3.7/dist-packages (from tensorflow==2.2.0) (1.30.0)
Requirement already satisfied: gast==0.3.3 in /usr/local/lib/python3.7/dist-packages (from tensorflow==2.2.0) (0.3.3)
Requirement already satisfied: scipy==1.4.1; python_version >= "3" in /usr/local/lib/python3.7/dist-packages (from tensorflow==2.2.0) (1.4.1)
Requirement already satisfied: tensorflow-estimator<2.3.0,>=2.2.0 in /usr/local/lib/python3.7/dist-packages (from tensorflow==2.2.0) (2.2.0)
Requirement already satisfied: h5py<2.11.0,>=2.10.0 in /usr/local/lib/python3.7/dist-packages (from tensorflow==2.2.0) (2.10.0)
Requirement already satisfied: setuptools in /usr/local/lib/python3.7/dist-packages (from protobuf>=3.8.0->tensorflow==2.2.0) (49.1.0)
Requirement already satisfied: markdown>=2.6.8 in /usr/local/lib/python3.7/dist-packages (from tensorboard<2.3.0,>=2.2.0->tensorflow==2.2.0) (3.2.2)
Requirement already satisfied: werkzeug>=0.11.15 in /usr/lib/python3/dist-packages (from tensorboard<2.3.0,>=2.2.0->tensorflow==2.2.0) (0.14.1)
Requirement already satisfied: tensorboard-plugin-wit>=1.6.0 in /usr/local/lib/python3.7/dist-packages (from tensorboard<2.3.0,>=2.2.0->tensorflow==2.2.0) (1.7.0)
Requirement already satisfied: requests<3,>=2.21.0 in /usr/lib/python3/dist-packages (from tensorboard<2.3.0,>=2.2.0->tensorflow==2.2.0) (2.21.0)
Requirement already satisfied: google-auth<2,>=1.6.3 in /usr/local/lib/python3.7/dist-packages (from tensorboard<2.3.0,>=2.2.0->tensorflow==2.2.0) (1.18.0)
Requirement already satisfied: google-auth-oauthlib<0.5,>=0.4.1 in /usr/local/lib/python3.7/dist-packages (from tensorboard<2.3.0,>=2.2.0->tensorflow==2.2.0) (0.4.1)
Requirement already satisfied: importlib-metadata; python_version < "3.8" in /usr/local/lib/python3.7/dist-packages (from markdown>=2.6.8->tensorboard<2.3.0,>=2.2.0->tensorflow==2.2.0) (1.7.0)
Requirement already satisfied: cachetools<5.0,>=2.0.0 in /usr/local/lib/python3.7/dist-packages (from google-auth<2,>=1.6.3->tensorboard<2.3.0,>=2.2.0->tensorflow==2.2.0) (4.1.1)
Requirement already satisfied: pyasn1-modules>=0.2.1 in /usr/local/lib/python3.7/dist-packages (from google-auth<2,>=1.6.3->tensorboard<2.3.0,>=2.2.0->tensorflow==2.2.0) (0.2.8)
Requirement already satisfied: rsa<5,>=3.1.4; python_version >= "3" in /usr/local/lib/python3.7/dist-packages (from google-auth<2,>=1.6.3->tensorboard<2.3.0,>=2.2.0->tensorflow==2.2.0) (4.6)
Requirement already satisfied: requests-oauthlib>=0.7.0 in /usr/lib/python3/dist-packages (from google-auth-oauthlib<0.5,>=0.4.1->tensorboard<2.3.0,>=2.2.0->tensorflow==2.2.0) (1.0.0)
Requirement already satisfied: zipp>=0.5 in /usr/local/lib/python3.7/dist-packages (from importlib-metadata; python_version < "3.8"->markdown>=2.6.8->tensorboard<2.3.0,>=2.2.0->tensorflow==2.2.0) (3.1.0)
Requirement already satisfied: pyasn1<0.5.0,>=0.4.6 in /usr/local/lib/python3.7/dist-packages (from pyasn1-modules>=0.2.1->google-auth<2,>=1.6.3->tensorboard<2.3.0,>=2.2.0->tensorflow==2.2.0) (0.4.8)
show this  
but when i try
$ python3
>>> import tensorflow

this eror showing
    import tensorflow
    ^
IndentationError: unexpected indent

hdihd9162, 2020-07-07:
and in mu 
Traceback (most recent call last):
  File "/home/pi/Desktop/mask.py", line 5, in <module>
    model = load_model('model-017.model')
  File "/usr/local/lib/python3.7/dist-packages/tensorflow/python/keras/saving/save.py", line 189, in load_model
    loader_impl.parse_saved_model(filepath)
  File "/usr/local/lib/python3.7/dist-packages/tensorflow/python/saved_model/loader_impl.py", line 113, in parse_saved_model
    constants.SAVED_MODEL_FILENAME_PB))
OSError: SavedModel file does not exist at: model-017.model/{saved_model.pbtxt|saved_model.pb}
this eror showing

pAP3R, 2020-07-07:
These are user issues, and not an issue with the Tensorflow project. 

Examine the error messages from python carefully, it is telling you what the problem is.

outrun7, 2020-07-07:
How do I use my computer?

hdihd9162, 2020-07-07:
> These are user issues, and not an issue with the Tensorflow project.
> 
> Examine the error messages from python carefully, it is telling you what the problem is.
this is my code can you fix it?
```
from keras.models import load_model
import cv2
import numpy as np

model = load_model('model-017.model')
face_clsfr=cv2.CascadeClassifier('haarcascade_frontalface_default.xml')

source=cv2.VideoCapture(0)

labels_dict={0:'with_mask',1:'without_mask'}
color_dict={0:(0,255,0),1:(0,0,255)}

while(True):

    ret,img=source.read()
    gray=cv2.cvtColor(img,cv2.COLOR_BGR2GRAY)
    faces=face_clsfr.detectMultiScale(gray,1.3,5)

    for x,y,w,h in faces:

        face_img=gray[y:y+w,x:x+w]
        resized=cv2.resize(face_img,(100,100))
        normalized=resized/255.0
        reshaped=np.reshape(normalized,(1,100,100,1))
        result=model.predict(reshaped)

        label=np.argmax(result,axis=1)[0]

        cv2.rectangle(img,(x,y),(x+w,y+h),color_dict[label],2)
        cv2.rectangle(img,(x,y-40),(x+w,y),color_dict[label],-1)
        cv2.putText(
          img, labels_dict[label],
          (x, y-10),
          cv2.FONT_HERSHEY_SIMPLEX,0.8,(255,255,255),2)

    cv2.imshow('LIVE',img)
    key=cv2.waitKey(1)

    if(key==27):
        break

cv2.destroyAllWindows()
source.release()
```

closed 2020-07-08 by hdihd9162