PINTO0309/Tensorflow-bin #26: Please provide FlexDelegate fix for arm71 in tf 2.3

stefan-lopez, 2020-07-12:
**[Required]** Your device (RaspberryPi3, LaptopPC, or other device name):  
  
**[Required]** Your device's CPU architecture (armv7l, x86_64, or other architecture name):  
  
**[Required]** Your OS (Raspbian, Ubuntu1604, or other os name):  
  
**[Required]** Details of the work you did before the problem occurred:  
  
**[Required]** Error message:  
  
**[Required]** Overview of problems and questions:

PINTO0309, 2020-07-12:
I've been trying to build it for 10 days because I want to use it myself, but TF v2.3.0 has been difficult and unsuccessful. I'm still working on my 20th build today. It's going to take a lot of time. If you're in a hurry, try the following repositories.

**Tool to compile tensorflow for ARM.**
**https://github.com/lhelontra/tensorflow-on-arm.git**

PINTO0309, 2020-07-13:
I have FlexDelegate enabled Tensorflow v2.3.0-rc1 committed armv7l's Wheel. The figure below shows my testing of the model with Flex OP and the inference results were returned successfully.
[image]

closed 2020-07-13 by PINTO0309