PINTO0309/Tensorflow-bin #27: Question: will XNNPACK compile flag be set in future builds?

filipvg68, 2020-07-24:
The TensorFlow/Google Research team just released following blog: https://blog.tensorflow.org/2020/07/accelerating-tensorflow-lite-xnnpack-integration.html, outlining a significant performance improvement with the XNNPACK library. For Linux, this would require to set an additional flag during the Bazel build: --define tflite_with_xnnpack=true

Is this something that will be considered to be included in future versions of your TF builds for the raspberry pi?

BTW - I'd like to thank you for your work to publish the TF builds for rpi!

Thanks,
Filip

PINTO0309, 2020-07-24:
I've already built and committed the package as a trial. However, I don't think you'll see much of a performance gain on ARM CPUs.
**https://github.com/PINTO0309/TensorflowLite-bin/tree/master/2.3.0-rc1**

filipvg68, 2020-07-25:
Thanks. As my application is using  tensorflow functionality that is not available in the tflite runtime, I've installed both tf and tflite_runtime, using the tflite_runtime for inference.

I did see a worse performance compared to the tf 2.2 build, but this likely because the interpreter.set_num_threads attribute is not enabled in the tflite_runtime build (I am using 4 threads for object detection on the raspberry pi 4).

Do you plan to make a build for the full tf 2.3 package with XNNPACK enabled?

PINTO0309, 2020-07-25:
>I did see a worse performance compared to the tf 2.2 build, but this likely because the interpreter.set_num_threads attribute is not enabled in the tflite_runtime build (I am using 4 threads for object detection on the raspberry pi 4).

Please read the instructions below carefully. Multi-Thread from TF v2.3.0 API usage has changed.
**https://github.com/PINTO0309/TensorflowLite-bin#note**
```python
interpreter = Interpreter(model_path="foo.tflite", num_threads=4)
```

>Do you plan to make a build for the full tf 2.3 package with XNNPACK enabled?

Yes. I'm working on it as a free hobby and will build it when I feel like it.

PINTO0309, 2020-07-28:
@filipvg68 
I first committed the Tensorflow v2.3.0 armv7l installer with FlexDelegate and XNNPACK enabled.
**https://github.com/PINTO0309/Tensorflow-bin/blob/master/tensorflow-2.3.0-cp37-none-linux_armv7l_download.sh**

closed 2020-07-28 by PINTO0309

filipvg68, 2020-07-28:
@PINTO0309 
Super, many thanks!