PINTO0309/Tensorflow-bin #28: TF-2.3.0-cp38-cp38m-linux_aarch64.whl is not supported wheel

Etoye, 2020-08-11:
**[Required]** Your device (RaspberryPi3, LaptopPC, or other device name):  RaspberryPi3 B+
  
**[Required]** Your device's CPU architecture (armv7l, x86_64, or other architecture name):  aarch64 
  
**[Required]** Your OS (Raspbian, Ubuntu1604, or other os name):  Ubuntu 20.04 64bit
  
**[Required]** Details of the work you did before the problem occurred:  Clean installation of Ubuntu 20.04
  
**[Required]** Error message:  tensorflow-2.3.0-cp38-cp38m-linux_aarch64.whl is not a supported wheel on this platform.
  
**[Required]** Overview of problems and questions:  Not able to install wheel due to error above, even if all conditions looks like correct.

Etoye, 2020-08-11:
Installed miniconda aarch64 with Python 3.7 and able to run tensorflow-2.3.0rc0-cp37-cp37m-linux_aarch64.whl 
Still not sure what is an issue with 3.8 thought.

PINTO0309, 2020-08-11:
Please rename the file name and try again.

From
**`tensorflow-2.3.0-cp38-cp38m-linux_aarch64.whl`**
To
**`tensorflow-2.3.0-cp38-none-linux_aarch64.whl`**

Etoye, 2020-08-11:
Installation started right after the file was renamed, thank you!

closed 2020-08-11 by Etoye