PINTO0309/Tensorflow-bin #33: Struggling with numpy version

StuartIanNaylor, 2021-05-17:
```
pip3 install tensorflow-2.5.0-cp37-none-linux_aarch64.whl
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Processing ./tensorflow-2.5.0-cp37-none-linux_aarch64.whl
Collecting opt-einsum~=3.3.0
  Downloading https://www.piwheels.org/simple/opt-einsum/opt_einsum-3.3.0-py3-none-any.whl (65 kB)
     |████████████████████████████████| 65 kB 654 kB/s 
Collecting tensorflow-estimator<2.6.0,>=2.5.0rc0
  Downloading tensorflow_estimator-2.5.0-py2.py3-none-any.whl (462 kB)
     |████████████████████████████████| 462 kB 113 kB/s 
Requirement already satisfied: absl-py~=0.10 in ./venv/lib/python3.7/site-packages (from tensorflow==2.5.0) (0.12.0)
Collecting google-pasta~=0.2
  Downloading https://www.piwheels.org/simple/google-pasta/google_pasta-0.2.0-py3-none-any.whl (57 kB)
     |████████████████████████████████| 57 kB 820 kB/s 
Collecting wheel~=0.35
  Downloading https://www.piwheels.org/simple/wheel/wheel-0.36.2-py2.py3-none-any.whl (35 kB)
Collecting keras-preprocessing~=1.1.2
  Downloading https://www.piwheels.org/simple/keras-preprocessing/Keras_Preprocessing-1.1.2-py2.py3-none-any.whl (42 kB)
     |████████████████████████████████| 42 kB 209 kB/s 
Collecting wrapt~=1.12.1
  Downloading wrapt-1.12.1.tar.gz (27 kB)
Collecting tensorboard~=2.5
  Downloading tensorboard-2.5.0-py3-none-any.whl (6.0 MB)
     |████████████████████████████████| 6.0 MB 123 kB/s 
Collecting termcolor~=1.1.0
  Downloading https://www.piwheels.org/simple/termcolor/termcolor-1.1.0-py3-none-any.whl (4.8 kB)
Collecting protobuf>=3.9.2
  Downloading protobuf-3.17.0-cp37-cp37m-manylinux2014_aarch64.whl (925 kB)
     |████████████████████████████████| 925 kB 374 kB/s 
Collecting flatbuffers~=1.12.0
  Downloading https://www.piwheels.org/simple/flatbuffers/flatbuffers-1.12-py2.py3-none-any.whl (15 kB)
Requirement already satisfied: numpy~=1.19.2 in ./venv/lib/python3.7/site-packages (from tensorflow==2.5.0) (1.19.5)
Collecting h5py~=3.1.0
  Downloading h5py-3.1.0.tar.gz (371 kB)
     |████████████████████████████████| 371 kB 799 kB/s 
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Installing backend dependencies ... done
    Preparing wheel metadata ... done
Collecting grpcio~=1.34.0
  Downloading grpcio-1.34.1.tar.gz (21.1 MB)
     |████████████████████████████████| 21.1 MB 23 kB/s 
Collecting typing-extensions~=3.7.4
  Downloading https://www.piwheels.org/simple/typing-extensions/typing_extensions-3.7.4.3-py3-none-any.whl (22 kB)
Collecting astunparse~=1.6.3
  Downloading https://www.piwheels.org/simple/astunparse/astunparse-1.6.3-py2.py3-none-any.whl (12 kB)
Collecting keras-nightly~=2.5.0.dev
  Downloading keras_nightly-2.5.0.dev2021032900-py2.py3-none-any.whl (1.2 MB)
     |████████████████████████████████| 1.2 MB 462 kB/s 
Collecting gast==0.4.0
  Downloading https://www.piwheels.org/simple/gast/gast-0.4.0-py3-none-any.whl (9.9 kB)
Collecting six~=1.15.0
  Downloading https://www.piwheels.org/simple/six/six-1.15.0-py2.py3-none-any.whl (10 kB)
Collecting cached-property
  Downloading https://www.piwheels.org/simple/cached-property/cached_property-1.5.2-py2.py3-none-any.whl (7.6 kB)
Collecting google-auth<2,>=1.6.3
  Downloading https://www.piwheels.org/simple/google-auth/google_auth-1.30.0-py2.py3-none-any.whl (146 kB)
     |████████████████████████████████| 146 kB 345 kB/s 
Collecting google-auth-oauthlib<0.5,>=0.4.1
  Downloading https://www.piwheels.org/simple/google-auth-oauthlib/google_auth_oauthlib-0.4.4-py2.py3-none-any.whl (18 kB)
Requirement already satisfied: requests<3,>=2.21.0 in /usr/lib/python3/dist-packages (from tensorboard~=2.5->tensorflow==2.5.0) (2.21.0)
Collecting tensorboard-plugin-wit>=1.6.0
  Downloading tensorboard_plugin_wit-1.8.0-py3-none-any.whl (781 kB)
     |████████████████████████████████| 781 kB 415 kB/s 
Collecting werkzeug>=0.11.15
  Downloading https://www.piwheels.org/simple/werkzeug/Werkzeug-2.0.0-py3-none-any.whl (288 kB)
     |████████████████████████████████| 288 kB 555 kB/s 
Requirement already satisfied: setuptools>=41.0.0 in ./venv/lib/python3.7/site-packages (from tensorboard~=2.5->tensorflow==2.5.0) (56.2.0)
Collecting markdown>=2.6.8
  Downloading https://www.piwheels.org/simple/markdown/Markdown-3.3.4-py3-none-any.whl (97 kB)
[... 41 lines cut ...]
    Can't uninstall 'wheel'. No files were found to uninstall.
  Attempting uninstall: h5py
    Found existing installation: h5py 2.10.0
    Uninstalling h5py-2.10.0:
      Successfully uninstalled h5py-2.10.0
Successfully installed astunparse-1.6.3 cached-property-1.5.2 cachetools-4.2.2 flatbuffers-1.12 gast-0.4.0 google-auth-1.30.0 google-auth-oauthlib-0.4.4 google-pasta-0.2.0 grpcio-1.34.1 h5py-3.1.0 importlib-metadata-4.0.1 keras-nightly-2.5.0.dev2021032900 keras-preprocessing-1.1.2 markdown-3.3.4 oauthlib-3.1.0 opt-einsum-3.3.0 protobuf-3.17.0 pyasn1-0.4.8 pyasn1-modules-0.2.8 requests-oauthlib-1.3.0 rsa-4.7.2 six-1.15.0 tensorboard-2.5.0 tensorboard-data-server-0.6.1 tensorboard-plugin-wit-1.8.0 tensorflow-2.5.0 tensorflow-estimator-2.5.0 termcolor-1.1.0 typing-extensions-3.7.4.3 werkzeug-2.0.0 wheel-0.36.2 wrapt-1.12.1 zipp-3.4.1
(venv) pi@raspberrypi:~/g-kws $ python3
Python 3.7.3 (default, Jan 22 2021, 20:04:44) 
[GCC 8.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import tensorflow
RuntimeError: module compiled against API version 0xe but this version of numpy is 0xd
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/pi/g-kws/venv/lib/python3.7/site-packages/tensorflow/__init__.py", line 41, in <module>
    from tensorflow.python.tools import module_util as _module_util
  File "/home/pi/g-kws/venv/lib/python3.7/site-packages/tensorflow/python/__init__.py", line 40, in <module>
    from tensorflow.python.eager import context
  File "/home/pi/g-kws/venv/lib/python3.7/site-packages/tensorflow/python/eager/context.py", line 37, in <module>
    from tensorflow.python.client import pywrap_tf_session
  File "/home/pi/g-kws/venv/lib/python3.7/site-packages/tensorflow/python/client/pywrap_tf_session.py", line 23, in <module>
    from tensorflow.python._pywrap_tf_session import *
ImportError: SystemError: <built-in method __contains__ of dict object at 0x7f98e18630> returned a result with an error set

```

PINTO0309, 2021-05-17:
Is it difficult to upgrade the version of Numpy for reasons such as using it for production environments? I purposely built with **`numpy==1.20.3`** in order to optimize Cython's performance. 
```
$ sudo pip3 install numpy==1.20.3 --upgrade
```

What operating system are you using?

PINTO0309, 2021-05-17:
@StuartIanNaylor 
I have created a numpy==1.19.5 special version for you. I haven't checked it works, could you please check it works if you can?
https://github.com/PINTO0309/Tensorflow-bin/blob/main/tensorflow-2.5.0-cp37-none-linux_aarch64_numpy1195_download.sh

StuartIanNaylor, 2021-05-17:
Cause I can will do now, it was strange as install a later version and the dependencies of TF would kick it.
Will try it now.

Thanks.

StuartIanNaylor, 2021-05-17:
Wow that grpcio-1.34.1.tar.gz 4 hours pinto! :) poor pi3A+

I am just going to check some things and also not clear the cache so should still have that whl but install  pip3 install numpy==1.20.3 --upgrade and the orig you supplied.

I will close off with results

StuartIanNaylor, 2021-05-17:
Yeah pip3 install numpy==1.20.3 --upgrade
Then tensorflow 2.5 and it uninstalls and uses 1.19.5 from the cache which I don't want to purge due to grpcio

closed 2021-05-17 by StuartIanNaylor