PINTO0309/Tensorflow-bin #34: TF2.5 installation issue

filipvg68, 2021-05-17:
Does anyone know what could cause the installation of TF2.5 to force installing TF 2.4rc0 (see installation output below)?

Thanks, Filip

Processing ./tensorflow-2.5.0-cp38-none-linux_aarch64.whl
Requirement already satisfied: protobuf~=3.13.0 in ./python_virtual_envs/camera_env/lib/python3.8/site-packages (from tensorflow==2.5.0) (3.13.0)
Requirement already satisfied: grpcio~=1.32.0 in ./python_virtual_envs/camera_env/lib/python3.8/site-packages (from tensorflow==2.5.0) (1.32.0)
Requirement already satisfied: keras-preprocessing~=1.1.2 in ./python_virtual_envs/camera_env/lib/python3.8/site-packages (from tensorflow==2.5.0) (1.1.2)
Requirement already satisfied: google-pasta~=0.2 in ./python_virtual_envs/camera_env/lib/python3.8/site-packages (from tensorflow==2.5.0) (0.2.0)
Requirement already satisfied: six~=1.15.0 in ./python_virtual_envs/camera_env/lib/python3.8/site-packages (from tensorflow==2.5.0) (1.15.0)
Requirement already satisfied: wheel~=0.35 in ./python_virtual_envs/camera_env/lib/python3.8/site-packages (from tensorflow==2.5.0) (0.36.2)
Requirement already satisfied: absl-py~=0.10 in ./python_virtual_envs/camera_env/lib/python3.8/site-packages (from tensorflow==2.5.0) (0.12.0)
Requirement already satisfied: wrapt~=1.12.1 in ./python_virtual_envs/camera_env/lib/python3.8/site-packages (from tensorflow==2.5.0) (1.12.1)
Requirement already satisfied: termcolor~=1.1.0 in ./python_virtual_envs/camera_env/lib/python3.8/site-packages (from tensorflow==2.5.0) (1.1.0)
Requirement already satisfied: tensorboard~=2.3 in ./python_virtual_envs/camera_env/lib/python3.8/site-packages (from tensorflow==2.5.0) (2.4.1)
Requirement already satisfied: flatbuffers~=1.12.0 in ./python_virtual_envs/camera_env/lib/python3.8/site-packages (from tensorflow==2.5.0) (1.12)
Requirement already satisfied: opt-einsum~=3.3.0 in ./python_virtual_envs/camera_env/lib/python3.8/site-packages (from tensorflow==2.5.0) (3.3.0)
Requirement already satisfied: typing-extensions~=3.7.4 in ./python_virtual_envs/camera_env/lib/python3.8/site-packages (from tensorflow==2.5.0) (3.7.4.3)
Requirement already satisfied: astunparse~=1.6.3 in ./python_virtual_envs/camera_env/lib/python3.8/site-packages (from tensorflow==2.5.0) (1.6.3)
Requirement already satisfied: gast==0.3.3 in ./python_virtual_envs/camera_env/lib/python3.8/site-packages (from tensorflow==2.5.0) (0.3.3)
Requirement already satisfied: numpy~=1.19.2 in ./python_virtual_envs/camera_env/lib/python3.8/site-packages (from tensorflow==2.5.0) (1.19.5)
Requirement already satisfied: h5py~=2.10.0 in ./python_virtual_envs/camera_env/lib/python3.8/site-packages (from tensorflow==2.5.0) (2.10.0)
Requirement already satisfied: tensorflow-estimator~=2.3.0 in ./python_virtual_envs/camera_env/lib/python3.8/site-packages (from tensorflow==2.5.0) (2.3.0)
Requirement already satisfied: setuptools in ./python_virtual_envs/camera_env/lib/python3.8/site-packages (from protobuf~=3.13.0->tensorflow==2.5.0) (44.0.0)
Requirement already satisfied: werkzeug>=0.11.15 in ./python_virtual_envs/camera_env/lib/python3.8/site-packages (from tensorboard~=2.3->tensorflow==2.5.0) (1.0.1)
Requirement already satisfied: markdown>=2.6.8 in ./python_virtual_envs/camera_env/lib/python3.8/site-packages (from tensorboard~=2.3->tensorflow==2.5.0) (3.3.4)
Requirement already satisfied: google-auth<2,>=1.6.3 in ./python_virtual_envs/camera_env/lib/python3.8/site-packages (from tensorboard~=2.3->tensorflow==2.5.0) (1.27.1)
Requirement already satisfied: tensorboard-plugin-wit>=1.6.0 in ./python_virtual_envs/camera_env/lib/python3.8/site-packages (from tensorboard~=2.3->tensorflow==2.5.0) (1.8.0)
Requirement already satisfied: requests<3,>=2.21.0 in ./python_virtual_envs/camera_env/lib/python3.8/site-packages (from tensorboard~=2.3->tensorflow==2.5.0) (2.25.1)
Requirement already satisfied: google-auth-oauthlib<0.5,>=0.4.1 in ./python_virtual_envs/camera_env/lib/python3.8/site-packages (from tensorboard~=2.3->tensorflow==2.5.0) (0.4.3)
Requirement already satisfied: cachetools<5.0,>=2.0.0 in ./python_virtual_envs/camera_env/lib/python3.8/site-packages (from google-auth<2,>=1.6.3->tensorboard~=2.3->tensorflow==2.5.0) (4.2.1)
Requirement already satisfied: rsa<5,>=3.1.4 in ./python_virtual_envs/camera_env/lib/python3.8/site-packages (from google-auth<2,>=1.6.3->tensorboard~=2.3->tensorflow==2.5.0) (4.7.2)
Requirement already satisfied: pyasn1-modules>=0.2.1 in ./python_virtual_envs/camera_env/lib/python3.8/site-packages (from google-auth<2,>=1.6.3->tensorboard~=2.3->tensorflow==2.5.0) (0.2.8)
Requirement already satisfied: requests-oauthlib>=0.7.0 in ./python_virtual_envs/camera_env/lib/python3.8/site-packages (from google-auth-oauthlib<0.5,>=0.4.1->tensorboard~=2.3->tensorflow==2.5.0) (1.3.0)
Requirement already satisfied: pyasn1<0.5.0,>=0.4.6 in ./python_virtual_envs/camera_env/lib/python3.8/site-packages (from pyasn1-modules>=0.2.1->google-auth<2,>=1.6.3->tensorboard~=2.3->tensorflow==2.5.0) (0.4.8)
Requirement already satisfied: urllib3<1.27,>=1.21.1 in ./python_virtual_envs/camera_env/lib/python3.8/site-packages (from requests<3,>=2.21.0->tensorboard~=2.3->tensorflow==2.5.0) (1.26.3)
Requirement already satisfied: certifi>=2017.4.17 in ./python_virtual_envs/camera_env/lib/python3.8/site-packages (from requests<3,>=2.21.0->tensorboard~=2.3->tensorflow==2.5.0) (2020.12.5)
Requirement already satisfied: idna<3,>=2.5 in ./python_virtual_envs/camera_env/lib/python3.8/site-packages (from requests<3,>=2.21.0->tensorboard~=2.3->tensorflow==2.5.0) (2.10)
Requirement already satisfied: chardet<5,>=3.0.2 in ./python_virtual_envs/camera_env/lib/python3.8/site-packages (from requests<3,>=2.21.0->tensorboard~=2.3->tensorflow==2.5.0) (4.0.0)
Requirement already satisfied: oauthlib>=3.0.0 in ./python_virtual_envs/camera_env/lib/python3.8/site-packages (from requests-oauthlib>=0.7.0->google-auth-oauthlib<0.5,>=0.4.1->tensorboard~=2.3->tensorflow==2.5.0) (3.1.0)
Installing collected packages: tensorflow
Successfully installed tensorflow-2.4.0rc0

PINTO0309, 2021-05-17:
Try it.
```
$ sudo pip3 uninstall tensorboard-plugin-wit tb-nightly tensorboard \
tf-estimator-nightly tensorflow-gpu \
tensorflow tf-nightly tensorflow_estimator -y
```

```
$ sudo pip3 install tensorflow-2.5.0-cp38-none-linux_aarch64.whl
```

filipvg68, 2021-05-17:
Unfortunately, it installed again TF 2.4rc0:

(camera_env) ubuntu@ubuntu:~$ pip3 uninstall tensorboard-plugin-wit tb-nightly tensorboard \
> tf-estimator-nightly tensorflow-gpu \
> tensorflow tf-nightly tensorflow_estimator -y
Found existing installation: tensorboard-plugin-wit 1.8.0
Uninstalling tensorboard-plugin-wit-1.8.0:
  Successfully uninstalled tensorboard-plugin-wit-1.8.0
WARNING: Skipping tb-nightly as it is not installed.
Found existing installation: tensorboard 2.4.1
Uninstalling tensorboard-2.4.1:
  Successfully uninstalled tensorboard-2.4.1
WARNING: Skipping tf-estimator-nightly as it is not installed.
WARNING: Skipping tensorflow-gpu as it is not installed.
Found existing installation: tensorflow 2.4.0
Uninstalling tensorflow-2.4.0:
  Successfully uninstalled tensorflow-2.4.0
WARNING: Skipping tf-nightly as it is not installed.
Found existing installation: tensorflow-estimator 2.4.0
Uninstalling tensorflow-estimator-2.4.0:
  Successfully uninstalled tensorflow-estimator-2.4.0
(camera_env) ubuntu@ubuntu:~$ pip3 install tensorflow-2.5.0-cp38-none-linux_aarch64.whl
Processing ./tensorflow-2.5.0-cp38-none-linux_aarch64.whl
Requirement already satisfied: typing-extensions~=3.7.4 in ./python_virtual_envs/camera_env/lib/python3.8/site-packages (from tensorflow==2.5.0) (3.7.4.3)
Requirement already satisfied: absl-py~=0.10 in ./python_virtual_envs/camera_env/lib/python3.8/site-packages (from tensorflow==2.5.0) (0.12.0)
Requirement already satisfied: wheel~=0.35 in ./python_virtual_envs/camera_env/lib/python3.8/site-packages (from tensorflow==2.5.0) (0.36.2)
Requirement already satisfied: termcolor~=1.1.0 in ./python_virtual_envs/camera_env/lib/python3.8/site-packages (from tensorflow==2.5.0) (1.1.0)
Requirement already satisfied: h5py~=2.10.0 in ./python_virtual_envs/camera_env/lib/python3.8/site-packages (from tensorflow==2.5.0) (2.10.0)
Collecting tensorboard~=2.3
  Using cached tensorboard-2.5.0-py3-none-any.whl (6.0 MB)
Requirement already satisfied: wrapt~=1.12.1 in ./python_virtual_envs/camera_env/lib/python3.8/site-packages (from tensorflow==2.5.0) (1.12.1)
Requirement already satisfied: opt-einsum~=3.3.0 in ./python_virtual_envs/camera_env/lib/python3.8/site-packages (from tensorflow==2.5.0) (3.3.0)
Requirement already satisfied: google-pasta~=0.2 in ./python_virtual_envs/camera_env/lib/python3.8/site-packages (from tensorflow==2.5.0) (0.2.0)
Requirement already satisfied: keras-preprocessing~=1.1.2 in ./python_virtual_envs/camera_env/lib/python3.8/site-packages (from tensorflow==2.5.0) (1.1.2)
Requirement already satisfied: grpcio~=1.32.0 in ./python_virtual_envs/camera_env/lib/python3.8/site-packages (from tensorflow==2.5.0) (1.32.0)
Requirement already satisfied: flatbuffers~=1.12.0 in ./python_virtual_envs/camera_env/lib/python3.8/site-packages (from tensorflow==2.5.0) (1.12)
Requirement already satisfied: numpy~=1.19.2 in ./python_virtual_envs/camera_env/lib/python3.8/site-packages (from tensorflow==2.5.0) (1.19.5)
Requirement already satisfied: protobuf~=3.13.0 in ./python_virtual_envs/camera_env/lib/python3.8/site-packages (from tensorflow==2.5.0) (3.13.0)
Requirement already satisfied: astunparse~=1.6.3 in ./python_virtual_envs/camera_env/lib/python3.8/site-packages (from tensorflow==2.5.0) (1.6.3)
Requirement already satisfied: gast==0.3.3 in ./python_virtual_envs/camera_env/lib/python3.8/site-packages (from tensorflow==2.5.0) (0.3.3)
Requirement already satisfied: six~=1.15.0 in ./python_virtual_envs/camera_env/lib/python3.8/site-packages (from tensorflow==2.5.0) (1.15.0)
Collecting tensorflow-estimator~=2.3.0
  Using cached tensorflow_estimator-2.3.0-py2.py3-none-any.whl (459 kB)
Requirement already satisfied: setuptools in ./python_virtual_envs/camera_env/lib/python3.8/site-packages (from protobuf~=3.13.0->tensorflow==2.5.0) (44.0.0)
Requirement already satisfied: requests<3,>=2.21.0 in ./python_virtual_envs/camera_env/lib/python3.8/site-packages (from tensorboard~=2.3->tensorflow==2.5.0) (2.25.1)
Requirement already satisfied: markdown>=2.6.8 in ./python_virtual_envs/camera_env/lib/python3.8/site-packages (from tensorboard~=2.3->tensorflow==2.5.0) (3.3.4)
Requirement already satisfied: werkzeug>=0.11.15 in ./python_virtual_envs/camera_env/lib/python3.8/site-packages (from tensorboard~=2.3->tensorflow==2.5.0) (1.0.1)
Collecting tensorboard-data-server<0.7.0,>=0.6.0
  Downloading tensorboard_data_server-0.6.1-py3-none-any.whl (2.4 kB)
Collecting tensorboard-plugin-wit>=1.6.0
  Using cached tensorboard_plugin_wit-1.8.0-py3-none-any.whl (781 kB)
Requirement already satisfied: google-auth-oauthlib<0.5,>=0.4.1 in ./python_virtual_envs/camera_env/lib/python3.8/site-packages (from tensorboard~=2.3->tensorflow==2.5.0) (0.4.3)
Requirement already satisfied: google-auth<2,>=1.6.3 in ./python_virtual_envs/camera_env/lib/python3.8/site-packages (from tensorboard~=2.3->tensorflow==2.5.0) (1.27.1)
Requirement already satisfied: rsa<5,>=3.1.4 in ./python_virtual_envs/camera_env/lib/python3.8/site-packages (from google-auth<2,>=1.6.3->tensorboard~=2.3->tensorflow==2.5.0) (4.7.2)
Requirement already satisfied: cachetools<5.0,>=2.0.0 in ./python_virtual_envs/camera_env/lib/python3.8/site-packages (from google-auth<2,>=1.6.3->tensorboard~=2.3->tensorflow==2.5.0) (4.2.1)
Requirement already satisfied: pyasn1-modules>=0.2.1 in ./python_virtual_envs/camera_env/lib/python3.8/site-packages (from google-auth<2,>=1.6.3->tensorboard~=2.3->tensorflow==2.5.0) (0.2.8)
[... cut ...]
Requirement already satisfied: chardet<5,>=3.0.2 in ./python_virtual_envs/camera_env/lib/python3.8/site-packages (from requests<3,>=2.21.0->tensorboard~=2.3->tensorflow==2.5.0) (4.0.0)
Requirement already satisfied: oauthlib>=3.0.0 in ./python_virtual_envs/camera_env/lib/python3.8/site-packages (from requests-oauthlib>=0.7.0->google-auth-oauthlib<0.5,>=0.4.1->tensorboard~=2.3->tensorflow==2.5.0) (3.1.0)
Installing collected packages: tensorboard-plugin-wit, tensorboard-data-server, tensorflow-estimator, tensorboard, tensorflow
Successfully installed tensorboard-2.5.0 tensorboard-data-server-0.6.1 tensorboard-plugin-wit-1.8.0 tensorflow-2.4.0rc0 tensorflow-estimator-2.3.0

filipvg68, 2021-05-21:
When installing TF 2.5 via link provided in Python 3.9 issue: https://snapshots.linaro.org/ldcg/python/tensorflow-manylinux/11/ , it worked well. So, my current assumption is that the tensorflow-2.5.0-cp38-none-linux_aarch64 wheel may have an issue. Not sure if anyone else had a problem. If not, I'll close this issue.

PINTO0309, 2021-05-21:
Follow the issue template and describe the environment you are using.

```
$ uname -a
Linux ip-172-31-13-75 5.4.0-1048-aws #50-Ubuntu SMP Mon May 3 21:42:45 UTC 2021 aarch64 aarch64 aarch64 GNU/Linux

$ cat /etc/issue
Ubuntu 20.04.2 LTS \n \l

$ mkdir work && cd work
$ python3 -c 'import tensorflow as tf; print(tf.__version__)'
Traceback (most recent call last):
  File "<string>", line 1, in <module>
ModuleNotFoundError: No module named 'tensorflow'

$ wget "https://github.com/PINTO0309/Tensorflow-bin/raw/main/tensorflow-2.5.0-cp38-none-linux_aarch64_numpy1195_download.sh"
$ sudo chmod +x tensorflow-2.5.0-cp38-none-linux_aarch64_numpy1195_download.sh

$ ./tensorflow-2.5.0-cp38-none-linux_aarch64_numpy1195_download.sh
$ ls -l
-rw-rw-r-- 1 ubuntu ubuntu 262376996 May 21 11:44 tensorflow-2.5.0-cp38-none-linux_aarch64.whl
-rwxrwxr-x 1 ubuntu ubuntu       373 May 21 11:42 tensorflow-2.5.0-cp38-none-linux_aarch64_numpy1195_download.sh

$ sudo pip3 install tensorflow-2.5.0-cp38-none-linux_aarch64.whl

Processing ./tensorflow-2.5.0-cp38-none-linux_aarch64.whl
Collecting flatbuffers~=1.12.0
  Downloading flatbuffers-1.12-py2.py3-none-any.whl (15 kB)
Collecting wheel~=0.35
  Using cached wheel-0.36.2-py2.py3-none-any.whl (35 kB)
Collecting absl-py~=0.10
  Downloading absl_py-0.12.0-py3-none-any.whl (129 kB)
     |████████████████████████████████| 129 kB 31.8 MB/s 
Collecting protobuf>=3.9.2
  Downloading protobuf-3.17.0-cp38-cp38-manylinux2014_aarch64.whl (925 kB)
     |████████████████████████████████| 925 kB 68.6 MB/s 
Collecting termcolor~=1.1.0
  Downloading termcolor-1.1.0.tar.gz (3.9 kB)
Collecting opt-einsum~=3.3.0
  Downloading opt_einsum-3.3.0-py3-none-any.whl (65 kB)
     |████████████████████████████████| 65 kB 7.9 MB/s 
Collecting tensorflow-estimator<2.6.0,>=2.5.0rc0
  Downloading tensorflow_estimator-2.5.0-py2.py3-none-any.whl (462 kB)
     |████████████████████████████████| 462 kB 100.9 MB/s 
Requirement already satisfied: h5py~=3.1.0 in /usr/local/lib/python3.8/dist-packages (from tensorflow==2.5.0) (3.1.0)
Collecting typing-extensions~=3.7.4
  Downloading typing_extensions-3.7.4.3-py3-none-any.whl (22 kB)
Collecting grpcio~=1.34.0
  Downloading grpcio-1.34.1.tar.gz (21.1 MB)
     |████████████████████████████████| 21.1 MB 35.6 MB/s 
Collecting tensorboard~=2.5
  Downloading tensorboard-2.5.0-py3-none-any.whl (6.0 MB)
     |████████████████████████████████| 6.0 MB 85.5 MB/s 
Collecting six~=1.15.0
  Downloading six-1.15.0-py2.py3-none-any.whl (10 kB)
Collecting astunparse~=1.6.3
  Downloading astunparse-1.6.3-py2.py3-none-any.whl (12 kB)
Collecting keras-nightly~=2.5.0.dev
  Downloading keras_nightly-2.5.0.dev2021032900-py2.py3-none-any.whl (1.2 MB)
     |████████████████████████████████| 1.2 MB 92.4 MB/s 
Collecting keras-preprocessing~=1.1.2
  Downloading Keras_Preprocessing-1.1.2-py2.py3-none-any.whl (42 kB)
     |████████████████████████████████| 42 kB 100 kB/s 
Collecting gast==0.4.0
  Downloading gast-0.4.0-py3-none-any.whl (9.8 kB)
Collecting google-pasta~=0.2
  Downloading google_pasta-0.2.0-py3-none-any.whl (57 kB)
     |████████████████████████████████| 57 kB 231 kB/s 
Requirement already satisfied: numpy~=1.19.2 in /usr/local/lib/python3.8/dist-packages (from tensorflow==2.5.0) (1.19.5)
Collecting wrapt~=1.12.1
[... 33 lines cut ...]
  Created wheel for grpcio: filename=grpcio-1.34.1-cp38-cp38-linux_aarch64.whl size=37704170 sha256=38c44d258b189f59fa30a8fc436131ac5f5cabde3736198c011b9e2a4a84ed51
  Stored in directory: /root/.cache/pip/wheels/26/34/49/8b3cd6300ee77fe0f02fe9fb0548583a6f3adce856cc5d9046
  Building wheel for wrapt (setup.py) ... done
  Created wheel for wrapt: filename=wrapt-1.12.1-cp38-cp38-linux_aarch64.whl size=77888 sha256=556c85206c6467b172856fe4e9b050cdbeee62a1eac348f2969c62b0d642305e
  Stored in directory: /root/.cache/pip/wheels/5f/fd/9e/b6cf5890494cb8ef0b5eaff72e5d55a70fb56316007d6dfe73
Successfully built termcolor grpcio wrapt
ERROR: launchpadlib 1.10.13 requires testresources, which is not installed.
Installing collected packages: flatbuffers, wheel, six, absl-py, protobuf, termcolor, opt-einsum, tensorflow-estimator, typing-extensions, grpcio, tensorboard-data-server, rsa, cachetools, google-auth, requests-oauthlib, google-auth-oauthlib, tensorboard-plugin-wit, markdown, werkzeug, tensorboard, astunparse, keras-nightly, keras-preprocessing, gast, google-pasta, wrapt, tensorflow
  Attempting uninstall: wheel
    Found existing installation: wheel 0.34.2
    Not uninstalling wheel at /usr/lib/python3/dist-packages, outside environment /usr
    Can't uninstall 'wheel'. No files were found to uninstall.
  Attempting uninstall: six
    Found existing installation: six 1.14.0
    Not uninstalling six at /usr/lib/python3/dist-packages, outside environment /usr
    Can't uninstall 'six'. No files were found to uninstall.
  Attempting uninstall: keras-preprocessing
    Found existing installation: Keras-Preprocessing 1.1.0
    Uninstalling Keras-Preprocessing-1.1.0:
      Successfully uninstalled Keras-Preprocessing-1.1.0
Successfully installed absl-py-0.12.0 astunparse-1.6.3 cachetools-4.2.2 flatbuffers-1.12 gast-0.4.0 google-auth-1.30.0 google-auth-oauthlib-0.4.4 google-pasta-0.2.0 grpcio-1.34.1 keras-nightly-2.5.0.dev2021032900 keras-preprocessing-1.1.2 markdown-3.3.4 opt-einsum-3.3.0 protobuf-3.17.0 requests-oauthlib-1.3.0 rsa-4.7.2 six-1.15.0 tensorboard-2.5.0 tensorboard-data-server-0.6.1 tensorboard-plugin-wit-1.8.0 tensorflow-2.5.0 tensorflow-estimator-2.5.0 termcolor-1.1.0 typing-extensions-3.7.4.3 werkzeug-2.0.1 wheel-0.36.2 wrapt-1.12.1

$ python3 -c 'import tensorflow as tf; print(tf.__version__)'
2.5.0
```

closed 2021-05-21 by PINTO0309

emojifreak, 2021-05-22:
Hi @PINTO0309 The following is a minor comment. `pip3` in Raspberry Pi OS/Debian Buster is too old (version 18), and the installation instruction at https://github.com/PINTO0309/Tensorflow-bin/blob/main/README.md gives the following error:

```
# pip3 install tensorflow-2.5.0-cp37-none-linux_armv7l.whl
tensorflow-2.5.0-cp37-none-linux_armv7l.whl is not a supported wheel on this platform.
# pip3 --version
pip 18.1 from /usr/lib/python3/dist-packages/pip (python 3.7)
```

emojifreak, 2021-05-22:
Another minor comment: in "Example of Python 3.x + Tensorflow v2 series", `cython`  should be `cython3` and `python-dev` should be removed as `python3-dev` is already included...

emojifreak, 2021-05-22:
On Ubuntu Focal arm64, `pip3 install h5py==3.1.0` fails, while the same sequence of commands succeeds on Ubuntu Hirsute, and Debian Buster&Bullseye. I thought it is an issue in Ubuntu, and the details are reported at https://bugs.launchpad.net/pip/+bug/1929259

PINTO0309, 2021-05-22:
@emojifreak 

- Ubuntu 20.04 aarch64 (arm64)
```
$ sudo pip3 install h5py==3.1.0 --force-reinstall
Processing /root/.cache/pip/wheels/65/ed/5c/a578025366e19f50064a590b01f6841f5a020730b6ea390a0e/h5py-3.1.0-cp38-cp38-linux_aarch64.whl
Collecting numpy>=1.17.5; python_version == "3.8"
  Using cached numpy-1.20.3-cp38-cp38-manylinux_2_17_aarch64.manylinux2014_aarch64.whl (12.7 MB)
ERROR: tensorflow 2.5.0 has requirement numpy~=1.19.2, but you'll have numpy 1.20.3 which is incompatible.
Installing collected packages: numpy, h5py
  Attempting uninstall: numpy
    Found existing installation: numpy 1.20.3
    Uninstalling numpy-1.20.3:
      Successfully uninstalled numpy-1.20.3
  Attempting uninstall: h5py
    Found existing installation: h5py 3.1.0
    Uninstalling h5py-3.1.0:
      Successfully uninstalled h5py-3.1.0
Successfully installed h5py-3.1.0 numpy-1.20.3
```

emojifreak, 2021-05-22:
@PINTO0309 Thanks for your response. `--force-reinstall` gave the same symptom.

```
Linux arm64-focal 5.10.38-rt39 #1 SMP PREEMPT_RT Thu May 20 10:35:43 JST 2021 aarch64 aarch64 aarch64 GNU/Linux
# cat /etc/issue
Ubuntu 20.04.2 LTS \n \l
```

This is a systemd container built by 

```
#!/bin/bash
SUITE=hirsute
ARCH=arm64
set -ex
apt-get install debian-ports-archive-keyring systemd-container
mmdebstrap --verbose --components="main restricted universe multiverse" --architectures=$ARCH --variant=standard \
 --include=curl,language-pack-en,build-essential,systemd-sysv,libpam-systemd,libnss-systemd,dbus-user-session,locales,tzdata,apt-utils,whiptail,vim-tiny,bash \
 $SUITE /var/lib/machines/${ARCH}-${SUITE}
cat >/var/lib/machines/${ARCH}-${SUITE}/etc/apt/apt.conf.d/00myconf <<EOF
APT::Default-Release "$SUITE";
APT::Install-Recommends 0;
APT::Get::Purge 1;
APT::Get::Upgrade-Allow-New 1;
Acquire::http::Proxy "http://192.168.1.2:8000/";
EOF
echo "${ARCH}-${SUITE}" >/var/lib/machines/${ARCH}-${SUITE}/etc/hostname
echo -e "127.0.1.1\t${ARCH}-${SUITE}" >>/var/lib/machines/${ARCH}-${SUITE}/etc/hosts
systemd-nspawn -q -M ${ARCH}-${SUITE} -a dpkg-reconfigure tzdata
systemd-nspawn -q -M ${ARCH}-${SUITE} -a dpkg-reconfigure locales
systemd-nspawn -q -M ${ARCH}-${SUITE} -a apt-get update
systemd-nspawn -q -M ${ARCH}-${SUITE} -a apt-get -y --autoremove install systemd-cron nullmailer popularity-contest unattended-upgrades
rm -f /var/lib/machines/${ARCH}-${SUITE}/etc/resolv.conf
cat >/var/lib/machines/${ARCH}-${SUITE}/etc/resolv.conf <<EOF
nameserver 192.168.1.2
options inet6 edns0
EOF
```

My main OS is Debian Bullseye arm64, which is the host OS running different containers.

emojifreak, 2021-05-23:
`CC=mpicc CXX=mpic++ pip3 install h5py==3.1.0` succeeds on Ubuntu 20.04 Focal with no error.
It is unclear why/how only Focal needs `CC=mpicc CXX=mpic++`.

PINTO0309, 2021-05-23:
Hmm, I don't understand why the behavior is different with systemd containers.