PINTO0309/Tensorflow-bin #35: Python 3.9 build

emojifreak, 2021-05-18:
**[Required]** Your device (RaspberryPi3, LaptopPC, or other device name):  Raspberry Pi 4B
  
**[Required]** Your device's CPU architecture (armv7l, x86_64, or other architecture name):  aarch64
  
**[Required]** Your OS (Raspbian, Ubuntu1604, or other os name):  Debian 11 Bullseye
  
**[Required]** Details of the work you did before the problem occurred:  There is no build for python 3.9
    
**[Required]** Overview of problems and questions:  
  
Ubuntu Hirsute
https://packages.ubuntu.com/hirsute/python3
and Debian 11 Bullseye
https://packages.debian.org/ja/bullseye/python3
now have python3.9 as the default python. Is there any plan to build binaries for python 3.9?

PINTO0309, 2021-05-18:
Can you explain the magnitude of the benefit of supporting a non-LTS OS? I don't want to create installers for non-LTS operating systems if possible because my maintenance costs will be very large.

Is there a particular reason, for example, that it is a very important part of the product, or that it has a special function that can only be achieved with this OS? If you have any interesting information that is worth my future significant maintenance costs, please provide it. :smiley:

emojifreak, 2021-05-18:
Debian Bullseye is LTS, as far as I know...

PINTO0309, 2021-05-18:
Python3.9 only supports TensorFlow v2.6.0 or tf-nightly. v2.5.0 will cause build errors. I don't know if this will be helpful, but I will describe a simple cross-compilation procedure using Docker. It will take a few hours to complete the build. I just started the build, but I don't know if it will finish successfully, and tf-nighty is not supported by me regardless of OS LTS.

- Debian bullseye aarch64 emulation by x86 host (tensorflow main branch build)
https://zenn.dev/pinto0309/scraps/b89dce1d5349fd

PINTO0309, 2021-05-18:
The build failed with the same error as this issue.
https://github.com/tensorflow/tensorflow/issues/49136

emojifreak, 2021-05-18:
@PINTO0309 Thank you very much for build trial. TF 2.5 claims support of Python 3.9, and errors with python 3.9 are tracked at https://github.com/tensorflow/tensorflow/issues/44485 
The build error with Aarch64 could be included in the above ticket... (I am unsure...)

emojifreak, 2021-05-18:
@PINTO0309 Thank you very much!
The builds with Python 3.7 and 3.8 seem successful (as you provide their binaries), this build error could also be mentioned at https://github.com/tensorflow/tensorflow/issues/44485 (I am unsure...)

hrw, 2021-05-19:
When it comes to released versions we have this coverage on AArch64 builds (using manylinux2014 image):

TF version | Py 3.6 |  Py 3.7 | Py 3.8 | Py 3.9
-|-|-|-|-
1.15.5 | OK | OK | -- | --
2.4.1 | OK | OK | OK | OK
2.5 | OK | OK | OK | OK

https://snapshots.linaro.org/ldcg/python/tensorflow-manylinux/4/tensorflow_cpu/ for 1.15.5 and 2.4.1 wheels
https://snapshots.linaro.org/ldcg/python/tensorflow-manylinux/11/tensorflow_cpu/ for 2.5.0 ones.

git HEAD is what fails on any Python version

PINTO0309, 2021-05-19:
@hrw 
Thank you for sharing this great information! The following link seems to be Page not found, is it under maintenance?
https://snapshots.linaro.org/ldcg/python/tensorflow-manylinux/11/tensorflow_cpu/

hrw, 2021-05-19:
fixed links

PINTO0309, 2021-05-19:
@emojifreak 
I've successfully compiled it and will share it with you.
- Debian 11 + Python3.9 + FlexDelegate + XNNPACK + Numpy 1.20.3
https://github.com/PINTO0309/Tensorflow-bin/blob/main/tensorflow-2.5.0-cp39-none-linux_aarch64_download.sh

emojifreak, 2021-05-20:
Thank you very much! It is time to close this issue.

closed 2021-05-20 by emojifreak