PINTO0309/Tensorflow-bin #37: Get Rrror Message When Import Tensorflow 2.5

hanifizzudinrahman, 2021-06-20:
**[Required]** Your device :  Raspberry Pi 3
  
**[Required]** Your device's CPU architecture :  armv7l
  
**[Required]** Your OS :  Raspbian i think
  
**[Required]** Details of the work you did before the problem occurred: I want to install tensorflow 2.5, and after install it (there is tensorflow 2.5 when i call "pip3 list"), there is error message, so i can'nt import it
  
**[Required]** Error message:  >>> import tensorflow
RuntimeError: module compiled against API version 0xe but this version of numpy is 0xd
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python3.7/dist-packages/tensorflow/__init__.py", line 41, in <module>
    from tensorflow.python.tools import module_util as _module_util
  File "/usr/local/lib/python3.7/dist-packages/tensorflow/python/__init__.py", line 40, in <module>
    from tensorflow.python.eager import context
  File "/usr/local/lib/python3.7/dist-packages/tensorflow/python/eager/context.py", line 37, in <module>
    from tensorflow.python.client import pywrap_tf_session
  File "/usr/local/lib/python3.7/dist-packages/tensorflow/python/client/pywrap_tf_session.py", line 23, in <module>
    from tensorflow.python._pywrap_tf_session import *
ImportError: SystemError: <built-in method __contains__ of dict object at 0x6fc25840> returned a result with an error set
>>> 
[image]

  
**[Required]** Overview of problems and questions:  How i solve this problem?
i follow this tutorial to install tensorflow in my raspberry pi 3
[image]

My future plan is want to make custom object detection with tensorflow lite in my raspberry pi 3

PINTO0309, 2021-06-20:
sudo pip3 install numpy --upgrade

hanifizzudinrahman, 2021-06-20:
[image]

Thank you brother, after i upgrade my numpy, it solved
thankyou very much

closed 2021-06-20 by PINTO0309