PINTO0309/Tensorflow-bin #38: Link is invalid

yuanqinong, 2021-08-16:
Hello, I try the follow command **$ wget "https://raw.githubusercontent.com/PINTO0309/Tensorflow-bin/master/tensorflow-1.15.0-cp37-cp37m-linux_armv7l_download.sh"** to download the TensorFlow but it shows an error 404 to me. Could help me about this issue?

`$ sudo apt-get install -y libhdf5-dev libc-ares-dev libeigen3-dev gcc gfortran libgfortran5 \
                          libatlas3-base libatlas-base-dev libopenblas-dev libopenblas-base libblas-dev \
                          liblapack-dev cython3 openmpi-bin libopenmpi-dev libatlas-base-dev python3-dev
$ sudo pip3 install pip --upgrade
$ sudo pip3 install keras_applications==1.0.8 --no-deps
$ sudo pip3 install keras_preprocessing==1.1.0 --no-deps
$ sudo pip3 install h5py==2.9.0
$ sudo pip3 install pybind11
$ pip3 install -U --user six wheel mock
$ sudo pip3 uninstall tensorflow
$ wget "https://raw.githubusercontent.com/PINTO0309/Tensorflow-bin/master/tensorflow-1.15.0-cp37-cp37m-linux_armv7l_download.sh"
$ ./tensorflow-1.15.0-cp37-cp37m-linux_armv7l_download.sh
$ sudo pip3 install tensorflow-1.15.0-cp37-cp37m-linux_armv7l.whl

【Required】 Restart the terminal.
`

PINTO0309, 2021-08-16:
It's a problem with your network environment. Unfortunately, there is nothing I can do.
```bash
~$ cd test
~/test$ wget "https://raw.githubusercontent.com/PINTO0309/Tensorflow-bin/master/tensorflow-1.15.0-cp37-cp37m-linux_armv7l_download.sh"
--2021-08-16 20:13:10--  https://raw.githubusercontent.com/PINTO0309/Tensorflow-bin/master/tensorflow-1.15.0-cp37-cp37m-linux_armv7l_download.sh
Resolving raw.githubusercontent.com (raw.githubusercontent.com)... 185.199.109.133, 185.199.110.133, 185.199.111.133, ...
Connecting to raw.githubusercontent.com (raw.githubusercontent.com)|185.199.109.133|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 374 [text/plain]
Saving to: ‘tensorflow-1.15.0-cp37-cp37m-linux_armv7l_download.sh’

tensorflow-1.15.0-c 100%[================>]     374  --.-KB/s    in 0s      

2021-08-16 20:13:10 (40.9 MB/s) - ‘tensorflow-1.15.0-cp37-cp37m-linux_armv7l_download.sh’ saved [374/374]

~/test$ ls -l
total 4
-rw-rw-r-- 1 xxxxx xxxxx 374  8月 16 20:13 tensorflow-1.15.0-cp37-cp37m-linux_armv7l_download.sh
~/test$ sudo chmod +x tensorflow-1.15.0-cp37-cp37m-linux_armv7l_download.sh 
~/test$ ./tensorflow-1.15.0-cp37-cp37m-linux_armv7l_download.sh 
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100   408    0   408    0     0   1263      0 --:--:-- --:--:-- --:--:--  1263
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
100 63.4M    0 63.4M    0     0  7927k      0 --:--:--  0:00:08 --:--:-- 11.3M
Download finished.
~/test$ ls -l
total 65004
-rwxrwxr-x 1 xxxxx xxxxx      374  8月 16 20:13 tensorflow-1.15.0-cp37-cp37m-linux_armv7l_download.sh
-rw-rw-r-- 1 xxxxx xxxxx 66556874  8月 16 20:13 tensorflow-1.15.0-cp37-cp37m-linux_armv7l.whl

```

closed 2021-08-16 by PINTO0309

yuanqinong, 2021-08-16:
Sorry dude, I inserted the wrong link. I am gonna try to install TensorFlow 2. Cloud you check again with this [link ](url).

yuanqinong, 2021-08-16:
> It's a problem with your network environment. Unfortunately, there is nothing I can do.
> 
> ```shell
> ~$ cd test
> ~/test$ wget "https://raw.githubusercontent.com/PINTO0309/Tensorflow-bin/master/tensorflow-1.15.0-cp37-cp37m-linux_armv7l_download.sh"
> --2021-08-16 20:13:10--  https://raw.githubusercontent.com/PINTO0309/Tensorflow-bin/master/tensorflow-1.15.0-cp37-cp37m-linux_armv7l_download.sh
> Resolving raw.githubusercontent.com (raw.githubusercontent.com)... 185.199.109.133, 185.199.110.133, 185.199.111.133, ...
> Connecting to raw.githubusercontent.com (raw.githubusercontent.com)|185.199.109.133|:443... connected.
> HTTP request sent, awaiting response... 200 OK
> Length: 374 [text/plain]
> Saving to: ‘tensorflow-1.15.0-cp37-cp37m-linux_armv7l_download.sh’
> 
> tensorflow-1.15.0-c 100%[================>]     374  --.-KB/s    in 0s      
> 
> 2021-08-16 20:13:10 (40.9 MB/s) - ‘tensorflow-1.15.0-cp37-cp37m-linux_armv7l_download.sh’ saved [374/374]
> 
> ~/test$ ls -l
> total 4
> -rw-rw-r-- 1 xxxxx xxxxx 374  8月 16 20:13 tensorflow-1.15.0-cp37-cp37m-linux_armv7l_download.sh
> ~/test$ sudo chmod +x tensorflow-1.15.0-cp37-cp37m-linux_armv7l_download.sh 
> ~/test$ ./tensorflow-1.15.0-cp37-cp37m-linux_armv7l_download.sh 
>   % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
>                                  Dload  Upload   Total   Spent    Left  Speed
> 100   408    0   408    0     0   1263      0 --:--:-- --:--:-- --:--:--  1263
>   0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
>   0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
> 100 63.4M    0 63.4M    0     0  7927k      0 --:--:--  0:00:08 --:--:-- 11.3M
> Download finished.
> ~/test$ ls -l
> total 65004
> -rwxrwxr-x 1 xxxxx xxxxx      374  8月 16 20:13 tensorflow-1.15.0-cp37-cp37m-linux_armv7l_download.sh
> -rw-rw-r-- 1 xxxxx xxxxx 66556874  8月 16 20:13 tensorflow-1.15.0-cp37-cp37m-linux_armv7l.whl
> ```

Sorry dude, I inserted the wrong link. I am gonna try to install TensorFlow 2. Cloud you check again with this link https://raw.githubusercontent.com/PINTO0309/Tensorflow-bin/master/tensorflow-2.6.0-cp37-none-linux_armv7l_download.sh .
```
pi@raspberrypi:~ $ wget "https://raw.githubusercontent.com/PINTO0309/Tensorflow-bin/master/tensorflow-2.6.0-cp37-none-linux_armv7l_download.sh"
--2021-08-16 13:24:22--  https://raw.githubusercontent.com/PINTO0309/Tensorflow-bin/master/tensorflow-2.6.0-cp37-none-linux_armv7l_download.sh
Resolving raw.githubusercontent.com (raw.githubusercontent.com)... 185.199.109.133, 185.199.111.133, 185.199.108.133, ...
Connecting to raw.githubusercontent.com (raw.githubusercontent.com)|185.199.109.133|:443... connected.
HTTP request sent, awaiting response... 404 Not Found
2021-08-16 13:24:23 ERROR 404: Not Found.
```

PINTO0309, 2021-08-16:
It's just that it hasn't been built and committed yet.
[image]