PINTO0309/Tensorflow-bin #40: issue on wget "https://raw.githubusercontent.com/PINTO0309/Tensorflow-bin/master/tensorflow-2.6.0-cp37-none-linux_aarch64_download.sh"

jinghan713, 2021-11-02:
**[Required]** Your device (RaspberryPi3, LaptopPC, or other device name):  
  RaspberryPi3 B+
**[Required]** Your device's CPU architecture (armv7l, x86_64, or other architecture name):  
  armv7l
**[Required]** Your OS (Raspbian, Ubuntu1604, or other os name):  
  Raspbian
**[Required]** Details of the work you did before the problem occurred:  
  I tried $ wget "https://raw.githubusercontent.com/PINTO0309/Tensorflow-bin/master/tensorflow-2.6.0-cp37-none-linux_aarch64_download.sh" but it show error 404 can't find the path
**[Required]** Error message:  
  HTTP request sent, awaiting response... 404 Not Found
**[Required]** Overview of problems and questions:  
  Not sure why the path link can't found

PINTO0309, 2021-11-02:
wget "https://raw.githubusercontent.com/PINTO0309/Tensorflow-bin/main/tensorflow-2.6.0-cp37-none-linux_aarch64_numpy1195_download.sh"

closed 2021-11-02 by PINTO0309

jinghan713, 2021-11-02:
Hi the access for the tensorflow-2.6.0-cp37-none-linux_aarch64_numpy1195_download.sh is denied unable to proceed . can you open the access? Thanks a lot

PINTO0309, 2021-11-02:
The aarch64 installer will not work on armv7l operating systems, you will need to download the armv7l files.

jinghan713, 2021-11-02:
Hi, sorry for disturbing but I faced another issue although I had use the armv7l files but I face issue while running sudo-H pip3 install tensorflow-2.5.0-cp37-none-linux armv7l.whl. I had tried apt.get purge python3-wrapt but I can't run the apt.get command
[image]