PINTO0309/Tensorflow-bin #43: TF 2.7.0 fails on Raspberry Pi 4 64-OS

Qengineering, 2021-11-21:
### **Your device: Raspberry Pi 4**
### **Your device's CPU architecture: aarch64**
### **Your OS: Raspberry Pi OS Buster (64-bit)**
### **Details of the work you did before the problem occurred:**

- Follow the Usage: "Example of Python 3.x + Tensorflow v2 series" steps up to the wget step
- From the the wget step on, use the tensorflow-2.7.0-cp37-none-linux_aarch64_numpy1214_download.sh script instead

### **Error message:**
ERROR: Could not find a version that satisfies the requirement libclang>= 9.0.1 (from tensorflow)<br/>
[image]

**### Overview of problems and questions:**
Tensorflow 2.7 seems to depend on the libclang 9.0.1 package
That version doesn't seem to be available for arm versions (aarch64 and armv7l) via pi.
Only the unsuitable version 7 can be installed with `$ sudo apt-get install libclang`
What is the solution here? TF 2.6 only optionally depending on clang.

PINTO0309, 2021-11-22:
https://github.com/llvm/llvm-project/releases/download/llvmorg-9.0.1/clang+llvm-9.0.1-aarch64-linux-gnu.tar.xz

or 

[debian buster aarch64 emulation (on x86_64) - llvm-9, clang-9](https://zenn.dev/pinto0309/scraps/a7569668da2b30)

closed 2021-11-22 by PINTO0309

Qengineering, 2021-12-06:
How do I install the clang+llvm-9.0.1-aarch64-linux-gnu.tar.xz in such a way, the TensorFlow pip3 recognize the libclang?
Just unzipping will not do.