PINTO0309/Tensorflow-bin #44: Tensorflow with python3.9 for armhf/armv7l

dnlmlr, 2021-11-28:
### Issue Type

Feature Request, Others

### OS

Other

### OS architecture

armv7

### Hardware

RaspberryPi4, RaspberryPi3

### Description

I am trying to get tensorflow running on the newly released `Raspberry Pi OS Bullseye` in the 32 bit / armv7l / armhf edition. I need the 32 bit version due to compatibility reasons. Since this new release comes with `python3.9`, there are no matching wheels provided. All wheels using `python3.9` (`*-cp39-*`) are also `aarch64` only.

Is there a reason why those versions are incompatible with `armv7l`? 

And if not, could you consider adding a wheel that works with `python3.9` and `armv7l`?

### Relevant Log Output

```shell
-
```

PINTO0309, 2021-11-28:
The reason is that since TensorFlow v2.6.0, compiling with armv7l has become an error and there is no workaround. I am constantly trying to compile. If it is compilable, I build and commit it a few days after the latest version is released, so wheel not being committed is synonymous with not being able to build.

PINTO0309, 2021-11-28:
If you just need to run TensorFlowLite, you can find the Python 3.9 binary here.
https://github.com/PINTO0309/TensorflowLite-bin

dnlmlr, 2021-11-28:
Thanks for the answer and the note on tensorflow lite 👍

footing11, 2021-12-05:
Is it possible if you just compile any version of TensorFlow for armv7l and python 3.9? I'm having a big problem with installing TensorFlow on RPi4. It seems like any available compiled version on the internet is not supported by python 3.9, since I'm getting the error "... is not supported on this platform".
Do I miss a lot of features if I just install TensorFlow Lite on my Raspberry Pi and do the quantization/building model for tflite on another PC? (Sorry for this basic question)
Thank you in advance,

PINTO0309, 2021-12-05:
@footing11 

>Do I miss a lot of features if I just install TensorFlow Lite on my Raspberry Pi and do the quantization/building model for tflite on another PC? (Sorry for this basic question)

You need to explain what you want to do first to be able to answer accurately. 
1. Is inference the only thing you want to do with your RaspberryPi?
2. Want to use RaspberryPi for training?
3. Do you need access to Kafka, OpenCL, AWS, or GCP?
4. Do you need to use saved_model to do the serving?
5. Do you need to run the tflite converter on a RaspberryPi?
6. etc...

It is not easy to anticipate what you are demanding in my answer. If 1 is Yes, TensorFlow's full build binaries are unnecessarily overkill. Incidentally, I have never wanted to install a full build binary on my RaspberryPi.

PINTO0309, 2021-12-05:
I'm going to close this because I've strayed into a topic that is not the essence of the issue.

closed 2021-12-05 by PINTO0309

footing11, 2021-12-05:
Thank you for the answer. Now I am pretty sure that I do need the full binarry on my Pi especially because I don't want to do just the inference on my Pi. I'm not looking for a specific project to do, but rather setting up a platform to learn and build different projects. 
Best,